facebookresearch/habitat-sim #1: Quick installation notes and missing dependencies

maximecb, 2019-02-27:
Hello. I was just trying to follow the quick installation instructions and wanted to provide a little bit of feedback so that other users can have a smoother experience.

1. The install command should either use `sudo`, or be run with `--user`, otherwise it complains about permission issues on Linux. What worked for me:

`python3 setup.py install --headless --user`

2. There are some dependencies I needed to install manually in order to get the build to work. These should ideally be mentioned in the README:

```
# Needed for EGL support:
sudo apt-get install libgles2-mesa-dev

sudo apt-get install python-dev 
```

maximecb, 2019-02-27:
Sidenote: it would be nice if you could ship a few more basic scenes with the simulator for quick testing.

erikwijmans, 2019-02-27:
Thank you very much.  We will certainly update the README to help smooth over common issues as we figure out what those are over the coming weeks.

mathfac, 2019-03-01:
Hi @maximecb 
> Sidenote: it would be nice if you could ship a few more basic scenes with the simulator for quick testing. It's a bit anticlimactic to install the simulator, and then have to go sign up some license agreement and download SUNCG of Matterport 3D datasets just to see what the scenes look like.
In [readme](https://github.com/facebookresearch/habitat-sim#testing) we propose to load the test scenes http://dl.fbaipublicfiles.com/habitat/habitat-test-scenes.zip. Did you try those?

maximecb, 2019-03-01:
Yes but those are two single-room 3D scans. I’d like to see a 3D mesh of an office or apartment, ideally with moving or movable objects, something that showcases your 3D engine better.

mathfac, 2019-03-01:
Thank you for the clarification, we are considering including photorealistic open-sourced scenes in the test data. We still can't include any of Matterport 3D scenes, as that requires sign up license agreement.

Skylion007, 2020-10-13:
The combination of the new quickstart features and /or colabs with test scenes should solve this issue.

closed 2020-10-13 by Skylion007