facebookresearch/habitat-sim #2: build problemos

odellus, 2019-02-27:
## Steps to reproduce

  0. `git clone --recursive https://github.com/facebookresearch/habitat-sim`
  1. `cd habitat-sim && ./build.sh && cd ~`
  2. `echo "export PYTHONPATH=PYTHONPATH:/path/to/habitat-sim" >> ~/.bashrc` 
  3. `source ~/.bashrc && python3.5 -c "import habitat_sim"`

## Observed Results
```
 Traceback (most recent call last):
  File "/home/thomas/code/habitat-sim/habitat_sim/bindings/__init__.py", line 10, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ImportError: No module named 'habitat_sim._ext'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/thomas/code/habitat-sim/habitat_sim/bindings/dev_bindings.py", line 72, in <module>
    import habitat_sim_bindings
ImportError: /home/thomas/code/habitat-sim/build/esp/bindings/habitat_sim_bindings.so: undefined symbol: _Py_ZeroStruct

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/thomas/code/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .simulator import *
  File "/home/thomas/code/habitat-sim/habitat_sim/simulator.py", line 7, in <module>
    from habitat_sim.bindings import *
  File "/home/thomas/code/habitat-sim/habitat_sim/bindings/__init__.py", line 18, in <module>
    from .dev_bindings import *
  File "/home/thomas/code/habitat-sim/habitat_sim/bindings/dev_bindings.py", line 84, in <module>
    raise ImportError(msg)
ImportError: 
Could not import habitat sim bindings
Please follow the building instructions in the README
-----------------------------------------------------------

Found the following libraries

 Library: habitat_sim_bindings.so -- Supported Python Version: Unknown

Your python interpreter is version 35
Please re-build habitat sim with this version of python
```

## Expected Results

Like nothing. `python3.5 -c "import habitat_sim"` then just nada.

odellus, 2019-02-27:
Okay a quick `cmake -LAH` in `build` showed that it was trying to build against python2.7. 

So I added 
```
cmake_args+=('-DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3.5')
cmake_args+=('-DPYTHON_INCLUDE_DIR=/usr/include/python3.5')
```
to line 59 of [`build.sh`](https://github.com/facebookresearch/habitat-sim/blob/master/build.sh#L59). 

Rebuilding. Have a feeling this will fix it. Will update soon.

odellus, 2019-02-28:
okay that didn't work.

Seeing
```
-- Found PythonInterp: /usr/bin/python3.5 (found version "3.5.2") 
-- Bindings being generated for python at /usr/bin/python
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so
```

So I added.
```
cmake_args+=('-DPYTHON_LIBRARY:FILEPATH=/usr/lib/python3.5/config-3.5m-x86_64-linux-gnu/libpython3.5.so')

```
to `build.sh` but I'm still seeing
```
-- Found PythonInterp: /usr/bin/python3.5 (found version "3.5.2") 
-- Bindings being generated for python at /usr/bin/python
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PythonLibs: /usr/lib/python3.5/config-3.5m-x86_64-linux-gnu/libpython3.5.so
```

Pretty weird. How do I get it to generate python3.5 bindings? Y'all developing on macs? :joy:

I'm not ready to upgrade to Ubuntu 18 because I use ROCm so I can't get rid of python2.7 as default.

So my hack was to change `python` to `python3.5` on Line 97 of [`src/cmake/dependencies.cmake`](https://github.com/facebookresearch/habitat-sim/blob/master/src/cmake/dependencies.cmake#L97).

Now I see 
```
-- Found PythonInterp: /usr/bin/python3.5 (found version "3.5.2") 
-- Bindings being generated for python at /usr/bin/python3.5
-- Found PythonLibs: /usr/lib/python3.5/config-3.5m-x86_64-linux-gnu/libpython3.5.so
-- pybind11 v2.3.dev0
```
which is what I would expect. Will report back after build is finished.

odellus, 2019-02-28:
Yup that did it. PR? Just close it bc they're hacks?

erikwijmans, 2019-02-28:
We have been ubiquitously using conda or virtual environments for development.  In those, whatever the version of python you are using gets aliased to just python, so we haven't run into this!

Glad to see you found a hacky work-around :)

odellus, 2019-02-28:
Thanks for chipping in Erik. Never been crazy about venv personally.

closed 2019-02-28 by odellus