facebookresearch/habitat-sim #4: Incorrect agent state returned

rakeshshrestha31, 2019-03-16:
## Steps to reproduce

  1. Replace the file [src/esp/gfx/Viewer.cpp](https://github.com/facebookresearch/habitat-sim/blob/master/src/esp/gfx/Viewer.cpp) with the file [Viewer.txt](https://github.com/facebookresearch/habitat-sim/files/2973550/Viewer.txt) (Note: the file extension should be changed to cpp)
  2. Build the project
  3. Run the "viewer" executable on one of the habitat-test-scenes
  4. Rotate the camera view using the arrow keys

## Observed Results
  * The y-rotation computed from the cameraNode_->getRotation() is always zero while the one computed from cameraNode_->getAbsoluteTransformation() is correct. The link below shows the results. This link shows the result: [video](https://drive.google.com/open?id=1MXZKLwcvPzyQSHeC6aoifHSI2sMdCjFS)
   * Similar thing is observed when using the python binding [habitat_sim/simulator.py](https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/simulator.py). Steps to reproduce this is not included for brevity, but can be provided if needed.

## Expected Results

  * The two rotations (default rotation and transformation based rotation) should be the same

## Relevant Code

```
  std::cout
    << std::fixed
    << std::setprecision(2)
    << "default rotation: " << cameraNode_->getRotation().toRotationMatrix().eulerAngles(1, 0, 2).transpose() * 180 / M_PI << std::endl
    <<"matrix based rotation: " << quaternion.toRotationMatrix().eulerAngles(1, 0, 2).transpose() * 180 / M_PI << std::endl
    << std::endl;
```
Connection of controls to navmesh is disabled for pure teleoperation
```
if (false) { // pathfinder_->isLoaded()) {
```

erikwijmans, 2019-03-16:
This is due to how the agent and the camera are related to each-other.  The camera maintains a translation and rotation relative to the agent's body.  When the arrow keys are pressed, this actually rotates the agents body, which in turn rotates the camera, thus the camera's location relative to the agent's body isn't changed.

`getRotation()` returns the rotation of the camera relative to the agent, while `getAbsoluteTransformation()` returns the location of the camera in world coordinates.

It is currently on our list of "todos" to properly expose the camera's location and we will make sure to do that in world coordinates.

You mention a similar issue with `example.py`?  `agent_state.rotation` only captures the rotation of the agent, so if you lookUp or lookDown, this won't be reflected in the agents rotation, but the agents rotation about the +Y axis should be captured by `agent_state.rotation`.

rakeshshrestha31, 2019-03-17:
Thank you for the explanation. Didn't know that the camera and the agent were related in that way.

And yes, the issue with `example.py` is how you described. Additionally, the agent's vertical translation (along y-axis) is not captured in the python interface.

PR #6 (2019-04-18): Agent in python
## Motivation and Context

Moves implementation of the agent into python.  There is no performance reason why that all needs to be in c++, so python it is!

Changed a lot of the Eigen parts that touch python to `const Eigen::Ref<const T>` instead of `const T&` as that allows pybind11 to do a better job at sharing memory.

Some additional things moving to python allows for:

1.   Allows for the agents action space to be indexed by any hashable type.  The agent's action space is just a dictionary where the action name (previously only strings) points to an `ActionSpec`.  The acti [... cut]
closed 2019-04-18 by erikwijmans