facebookresearch/habitat-sim #5: Suggestions to implement an image-goal based navigation

srama2512, 2019-03-17:
Hi, I'm trying to extend the current tasks to include image-based target-driven navigation. I've been looking at the simulator and api. So far, my understanding is that there are sensors attached to agents and these are passed to the renderer for getting images. I see that there is a method to render based on cameras alone (instead of a sensor tied to an agent). 

https://github.com/facebookresearch/habitat-sim/blob/b11d6e7f3c948cbff3a40672ff213bd51c625fe4/src/esp/gfx/Renderer.cpp#L132

I'm not sure how this can be used in the python API to get renderings of a target-location though. Is there a way to define a task sensor that takes in a pose of the world and provides the image from that pose?

erikwijmans, 2019-04-03:
The functionality needed to do this exists, but it is hidden away in c++ right now (we are working on that).  Currently, I think the easiest way to do this would be to temporarily move the agent with 

https://github.com/facebookresearch/habitat-sim/blob/master/src/esp/bindings/bindings.cpp#L581

get the observations via

https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/simulator.py#L69

and then restore the agent to its previous location.

mathfac, 2019-04-04:
@srama2512 what a scenes dataset are you working with? 
Habitat-API was designed specifically for task definitions. You can check:
1. [Navigation task definition](https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/nav/nav_task.py).
2. [Embodied Question and Answering definition](https://github.com/facebookresearch/habitat-api/blob/master/habitat/tasks/eqa/eqa_task.py).

srama2512, 2019-04-04:
@erikwijmans - Ah, got it. Moving the agent temporarily seems like a solution for now. Thanks! 

@mathfac - I'm currently trying to use both MatterPort and SUNCG. If I understand correctly, I would just have to create ImageNavigationTask and create a TargetSensor right? I wasn't sure how I would render images from the TargetSensor though. 

In House3D, I was able to create an extra camera (one that is static throughout the episode) and obtain the same information. Is there a functionality to do that at the moment?

erikwijmans, 2019-04-06:
> In House3D, I was able to create an extra camera (one that is static throughout the episode) and obtain the same information. Is there a functionality to do that at the moment?

This functionality does exist, and it is somewhat exposed into python (although it is currently exposed in a dangerous way, there are some caveats about memory ownership that are quite different than normal memory ownership in python, we are working on fix this).  In #6 you can see how to create a new node in the scene graph: https://github.com/facebookresearch/habitat-sim/pull/6/files#diff-8a755bab394eaa22725cd457449d7108R62, create a PinholeCamera that uses the node for rendering: https://github.com/facebookresearch/habitat-sim/pull/6/files#diff-142edc24b058301997a8b0d8ab887f6dR134, and finally how to move that camera to an arbitrary location: https://github.com/facebookresearch/habitat-sim/pull/6/files#diff-142edc24b058301997a8b0d8ab887f6dR236

If the camera is attached to a child of the root node instead of a child of the agent, it won't move with the agent.

The renderer will happily take a PinholeCamera object and render from that view (although you will then have to read the result into an array like we do here: https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/simulator.py#L121)

srama2512, 2019-04-10:
Thank you for the detailed instructions! I will close this issue now.

closed 2019-04-10 by srama2512