facebookresearch/habitat-sim #13: setup.py does not work

danielgordon10, 2019-04-11:
## Steps to reproduce

  1. git clone habitat-sim blah blah
  2. run `python setup.py install --headless
  3. run `python` then `import habitat_sim`

## Observed Results
```
Traceback (most recent call last):
  File "/data/xkcd/habitat-sim/habitat_sim/bindings/__init__.py", line 10, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ModuleNotFoundError: No module named 'habitat_sim._ext'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/data/xkcd/habitat-sim/habitat_sim/bindings/dev_bindings.py", line 72, in <module>
    import habitat_sim_bindings
ModuleNotFoundError: No module named 'habitat_sim_bindings'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/example.py", line 9, in <module>
    import demo_runner as dr
  File "/data/xkcd/habitat-sim/examples/demo_runner.py", line 15, in <module>
    import habitat_sim
  File "/data/xkcd/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .simulator import *
  File "/data/xkcd/habitat-sim/habitat_sim/simulator.py", line 7, in <module>
    from habitat_sim.bindings import *
  File "/data/xkcd/habitat-sim/habitat_sim/bindings/__init__.py", line 18, in <module>
    from .dev_bindings import *
  File "/data/xkcd/habitat-sim/habitat_sim/bindings/dev_bindings.py", line 84, in <module>
    raise ImportError(msg)
ImportError: 
Could not import habitat sim bindings
Please follow the building instructions in the README
-----------------------------------------------------------

Could not find the expected bindings build folder: "/data/xkcd/habitat-sim/build/esp/bindings"
Make sure to not delete the build folder after building
```

Seems like those files don't exist. Installing through the "developer instructions" still works fine.
```bash
(habitat-env) xkcd@ava:/data/xkcd/habitat-sim$ ls build
bdist.linux-x86_64  lib.linux-x86_64-3.6  temp.linux-x86_64-3.6
```

## Expected Results

  * habitat_sim imports correctly

erikwijmans, 2019-04-11:
It seems like it is picking up `habitat_sim` from `/data/xkcd/habitat-sim` instead of wherever it got installed too.  Can you try `unset PYTHONPATH`?

danielgordon10, 2019-04-12:
It actually was installed at `/data/xkcd/habitat-sim` Do you think that's an issue?

erikwijmans, 2019-04-12:
It seems like you current have a conda env or virtualenv called `habitat-env` active, so `python setup.py install --headless` will install things into the `site_packages` folder in that environment.

We definitely have seen oddities of using setup.py when not in an conda or virtual env, so that could be the issue if I am miss-reading the shell prompt.

danielgordon10, 2019-04-12:
My full setup is miniconda env on a separate hard drive than the main one with the OS (which ran out of space). The conda env is called habitat-env and is located at `/data/xkcd/miniconda`. Other conda packages so far seemed to install correctly. I was able to run `setup.py` and everything compiled. Here is the final output from that
```
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/habitat_sim
copying build/lib.linux-x86_64-3.6/habitat_sim/__init__.py -> build/bdist.linux-x86_64/egg/habitat_sim
creating build/bdist.linux-x86_64/egg/habitat_sim/_ext
copying build/lib.linux-x86_64-3.6/habitat_sim/_ext/habitat_sim_bindings.cpython-36m-x86_64-linux-gnu.so -> build/bdist.linux-x86_64/egg/habitat_sim/_ext
copying build/lib.linux-x86_64-3.6/habitat_sim/simulator.py -> build/bdist.linux-x86_64/egg/habitat_sim
copying build/lib.linux-x86_64-3.6/habitat_sim/utils.py -> build/bdist.linux-x86_64/egg/habitat_sim
creating build/bdist.linux-x86_64/egg/habitat_sim/bindings
copying build/lib.linux-x86_64-3.6/habitat_sim/bindings/__init__.py -> build/bdist.linux-x86_64/egg/habitat_sim/bindings
copying build/lib.linux-x86_64-3.6/habitat_sim/bindings/modules.py -> build/bdist.linux-x86_64/egg/habitat_sim/bindings
copying build/lib.linux-x86_64-3.6/habitat_sim/bindings/dev_bindings.py -> build/bdist.linux-x86_64/egg/habitat_sim/bindings
byte-compiling build/bdist.linux-x86_64/egg/habitat_sim/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/habitat_sim/simulator.py to simulator.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/habitat_sim/utils.py to utils.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/habitat_sim/bindings/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/habitat_sim/bindings/modules.py to modules.cpython-36.pyc
byte-compiling build/bdist.linux-x86_64/egg/habitat_sim/bindings/dev_bindings.py to dev_bindings.cpython-36.pyc
creating stub loader for habitat_sim/_ext/habitat_sim_bindings.cpython-36m-x86_64-linux-gnu.so
byte-compiling build/bdist.linux-x86_64/egg/habitat_sim/_ext/habitat_sim_bindings.py to habitat_sim_bindings.cpython-36.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying habitat_sim.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying habitat_sim.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying habitat_sim.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying habitat_sim.egg-info/not-zip-safe -> build/bdist.linux-x86_64/egg/EGG-INFO
copying habitat_sim.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying habitat_sim.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
writing build/bdist.linux-x86_64/egg/EGG-INFO/native_libs.txt
creating 'dist/habitat_sim-0.0.1-py3.6-linux-x86_64.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing habitat_sim-0.0.1-py3.6-linux-x86_64.egg
removing '/data/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-linux-x86_64.egg' (and everything under it)
creating /data/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-linux-x86_64.egg
Extracting habitat_sim-0.0.1-py3.6-linux-x86_64.egg to /data/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages
habitat-sim 0.0.1 is already the active version in easy-install.pth

Installed /data/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-linux-x86_64.egg
Processing dependencies for habitat-sim==0.0.1
Searching for numpy==1.16.2
Best match: numpy 1.16.2
Adding numpy 1.16.2 to easy-install.pth file
Installing f2py script to /data/xkcd/miniconda3/envs/habitat-env/bin
Installing f2py3 script to /data/xkcd/miniconda3/envs/habitat-env/bin
Installing f2py3.6 script to /data/xkcd/miniconda3/envs/habitat-env/bin

Using /data/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages
Finished processing dependencies for habitat-sim==0.0.1
```
The constructed build folder has three subdirectories: `bdist.linux-x86_64  lib.linux-x86_64-3.6  temp.linux-x86_64-3.6` but it looks like the bindings are expected to be in `build/esp/bindings` so I think there's an issue of either missing some files, or not updating the path to point to where it should be looking. Those files do of course exist when using the `build.sh` script, and I believe they are probably in one of the subfolders in the currently outputted build directory, but I don't know where. For example
```
(habitat-env) xkcd@ava:/data/xkcd/habitat-sim$ ls build/lib.linux-x86_64-3.6/habitat_sim/
bindings  _ext  __init__.py  simulator.py  utils.py
```

danielgordon10, 2019-04-12:
Solution found: instead of running 
`python setup.py install --headless`
I ran:
`pip install -e .`
like it says to in habitat-api, and that worked. So either the instructions in the readme should be updated, or pip install is copying some file that setup.py isn't to the lib location where it expects. That would be my guess at least.

erikwijmans, 2019-04-12:
Yeah, it's probably that your `PYTHONPATH` has `/data/xkcd/habitat-sim` in it.  

We have two different methods for building things, `setup.py` vs. `build.sh`.  Unfortunately the two different methods have to produce bindings in two different locations and thus there are two different code paths that look for them.  The code at first tries to load the bindings installed with `setup.py`, that's this error:
```
File "/data/xkcd/habitat-sim/habitat_sim/bindings/__init__.py", line 10, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ModuleNotFoundError: No module named 'habitat_sim._ext'
```

If it fails to find that, it looks for them in` build/esp/bindings`, the location `./build.sh` puts them in.

I should probably have `setup.py` set some value so I know which way it was built and I can reduce this confusion.

erikwijmans, 2019-04-12:
`pip install -e .` adds your current directory to the python's search path, so it probably using bindings built with `./build.sh`

danielgordon10, 2019-04-12:
Not exactly sure what you mean by "your PYTHONPATH has /data/xkcd/habitat-sim in it" It is not there. Should it be explicitly added? The instructions in the readme say you don't need to using the setup.py method, so if so they should be changed.
```
(habitat-env) xkcd@ava:/data/xkcd/habitat-sim$ echo $PYTHONPATH
:/home/xkcd/Documents/external_code/caffe/python:/home/xkcd/Documents/external_code/darknet/python/:/local/targaryen/Documents/workspace/rse_lab_useful_scripts/python/:/local/targaryen/Documents/external_code/tensorflow_models/:/local/targaryen/Documents/external_code/tensorflow_models/slim/:/local/targaryen/Documents/external_code/tensorflow_models/research/:/local/targaryen/Documents/external_code/tensorflow_models/research/slim/:/home/xkcd/Documents/external_code/selective_search_py/:/home/xkcd/Documents/external_code/vispy/:/home/xkcd/Documents/external_code/mxnet/python/:/local/targaryen/Documents/workspace/EmbodiedQA/House3D/
```

danielgordon10, 2019-04-12:
From the README

> For use with Habitat-API and your own python code, add habitat-sim to your PYTHONPATH (not necessary if you used python setup.py install instead of build.sh). For example modify your .bashrc (or .bash_profile in Mac osx) file by adding the line:

erikwijmans, 2019-04-12:
Are you doing something like

```
python
>>> import habitat_sim
```

while in `/data/xkcd/habitat-sim`?

erikwijmans, 2019-04-12:
As that'll pick up the package from your current directory. :/

danielgordon10, 2019-04-12:
...yes. BUT in my defense, the first thing I tried was `python examples/examples.py` which has the exact same error (and is also in the readme). I also could swear I tried importing from a different directory just to make sure and that it didn't work, but perhaps not.

erikwijmans, 2019-04-12:
Huh, that is odd.  `python examples/examples.py` _should_ not pick it up as `/data/xkcd/habitat-sim/examples` gets added to the path, not `/data/xkcd/habitat-sim`.  I will try to repro and I will definitely make the error message less confusing by not using a hacky way to determining how it was built

danielgordon10, 2019-04-17:
Let's call this fixed by https://github.com/facebookresearch/habitat-sim/pull/14

closed 2019-04-17 by danielgordon10