facebookresearch/habitat-sim #16: Build failed: could not find from_json() method in T's namespace

titardrew, 2019-04-14:
Build fails on:
* `OSX 10.14.3`
* both `gcc/g++ 4.2.1` and `clang/clang++ 10.0.1`

## Steps to reproduce

  1. `python setup.py install`

## Observed Results

```
[ 65%] Built target MagnumPrimitives
[ 65%] Compiling data resource file /Users/andriitytarenko/devel/habitat-root/habitat-sim/build/temp.macosx-10.14-x86_64-3.7/deps/magnum-plugins/src/MagnumPlugins/TinyGltfImporter/resource_TinyGltfImporter.cpp
Scanning dependencies of target TinyGltfImporter
[ 65%] Building CXX object deps/magnum-plugins/src/MagnumPlugins/TinyGltfImporter/CMakeFiles/TinyGltfImporter.dir/TinyGltfImporter.cpp.o
In file included from /Users/andriitytarenko/devel/habitat-root/habitat-sim/src/deps/magnum-plugins/src/MagnumPlugins/TinyGltfImporter/TinyGltfImporter.cpp:71:
In file included from /Users/andriitytarenko/devel/habitat-root/habitat-sim/src/deps/magnum-plugins/src/MagnumExternal/TinyGltf/tiny_gltf.h:992:
/Users/andriitytarenko/devel/habitat-root/habitat-sim/src/deps/magnum-plugins/src/MagnumExternal/TinyGltf/./json.hpp:1378:9: error: static_assert failed "could not find from_json() method in T's namespace"
        static_assert(sizeof(BasicJsonType) == 0,
        ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/andriitytarenko/devel/habitat-root/habitat-sim/src/deps/magnum-plugins/src/MagnumExternal/TinyGltf/./json.hpp:1387:16: note: in instantiation of function template specialization 'nlohmann::detail::from_json_fn::call<nlohmann::basic_json<std::map, std::vector,
      std::__1::basic_string<char>, bool, long long, unsigned long long, double, std::allocator, adl_serializer>, std::__1::basic_string_view<char, std::__1::char_traits<char> > >' requested here
        return call(j, val, priority_tag<1> {});
               ^
/Users/andriitytarenko/devel/habitat-root/habitat-sim/src/deps/magnum-plugins/src/MagnumExternal/TinyGltf/./json.hpp:6861:9: note: in instantiation of function template specialization 'nlohmann::detail::from_json_fn::operator()<nlohmann::basic_json<std::map, std::vector,
      std::__1::basic_string<char>, bool, long long, unsigned long long, double, std::allocator, adl_serializer>, std::__1::basic_string_view<char, std::__1::char_traits<char> > >' requested here
        ::nlohmann::from_json(std::forward<BasicJsonType>(j), val);
        ^
/Users/andriitytarenko/devel/habitat-root/habitat-sim/src/deps/magnum-plugins/src/MagnumExternal/TinyGltf/./json.hpp:9578:36: note: in instantiation of function template specialization 'nlohmann::adl_serializer<std::__1::basic_string_view<char, std::__1::char_traits<char>
      >, void>::from_json<const nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long long, unsigned long long, double, std::allocator, adl_serializer> &, std::__1::basic_string_view<char, std::__1::char_traits<char> > >' requested here
        JSONSerializer<ValueType>::from_json(*this, ret);
                                   ^
/Users/andriitytarenko/devel/habitat-root/habitat-sim/src/deps/magnum-plugins/src/MagnumExternal/TinyGltf/./json.hpp:9841:16: note: in instantiation of function template specialization 'nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long
      long, unsigned long long, double, std::allocator, adl_serializer>::get<std::__1::basic_string_view<char, std::__1::char_traits<char> >, std::__1::basic_string_view<char, std::__1::char_traits<char> >, 0>' requested here
        return get<ValueType>();
               ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/string:875:29: note: in instantiation of function template specialization 'nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long long, unsigned long long, double, std::allocator,
      adl_serializer>::operator basic_string_view<std::__1::basic_string_view<char, std::__1::char_traits<char> >, 0>' requested here
        {__self_view __sv = __t; return assign(__sv);}
                            ^
/Users/andriitytarenko/devel/habitat-root/habitat-sim/src/deps/magnum-plugins/src/MagnumExternal/TinyGltf/tiny_gltf.h:1924:12: note: in instantiation of function template specialization 'std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>
      >::operator=<nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long long, unsigned long long, double, std::allocator, adl_serializer>, void>' requested here
    (*ret) = it.value();
           ^
1 error generated.
make[2]: *** [deps/magnum-plugins/src/MagnumPlugins/TinyGltfImporter/CMakeFiles/TinyGltfImporter.dir/TinyGltfImporter.cpp.o] Error 1
make[1]: *** [deps/magnum-plugins/src/MagnumPlugins/TinyGltfImporter/CMakeFiles/TinyGltfImporter.dir/all] Error 2
```

It seems like it is known issue of MagnumPlugins (its TinyGltf's json dependency) e.g.:
https://github.com/mosra/magnum-plugins/issues/54
https://github.com/nlohmann/json/issues/539
https://github.com/nlohmann/json/issues/1140
https://github.com/nlohmann/json/issues/917

erikwijmans, 2019-04-15:
https://github.com/mosra/magnum-plugins/issues/54 says that the issue was fixed by updating to a newer version.  Can you try updating the magnum-plugins submodule (you may also need to update the magnum and corrade submodues) to confirm the fix?

titardrew, 2019-04-15:
@erikwijmans Well, I tried, but it did not help. The reason is that the commit which fixes this issue in magnum-plugins repo is more recent than latest releases in both magnum and corrade. In any case, even if that error disappears - another ones occur. **What really helped to fix my issue** is to make same changes as in commit that fixes https://github.com/mosra/magnum-plugins/issues/54 , so I just replaced my `json.hpp` with one from [commit](https://github.com/mosra/magnum-plugins/blob/b64e5b37e4cc2c3f95db16620ab1403378309d93/src/MagnumExternal/TinyGltf/json.hpp)
So I've just added
```
os.remove("src/deps/magnum-plugins/src/MagnumExternal/TinyGltf/json.hpp")

subprocess.call(
    ["wget", "https://raw.githubusercontent.com/mosra/magnum-plugins/b64e5b37e4cc2c3f95db16620ab1403378309d93/src/MagnumExternal/TinyGltf/json.hpp",
                "-P", "src/deps/magnum-plugins/src/MagnumExternal/TinyGltf"]
)
```
right after 
```
if is_in_git:
    subprocess.check_call(
        ["git", "submodule", "update", "--init", "--recursive"]
    )
```
in `build_extension` method of `setup.py` and it **now it works**.
Thanks for your help!

erikwijmans, 2019-04-15:
Glad you found a fix, we will update magnum in the not so distant future.  Keeping this open until then.

dhruvbatra, 2019-04-22:
Came here to report the same issue. We should really freeze the version/commit-id of magnum we use.

dhruvbatra, 2019-04-22:
Even after ^^this fix by @titardrew, I am seeing the following error with viewer: 

```
(base) dbatra-mbp:habitat-sim dbatra$ ./build/viewer ../data/scene_datasets/habitat-test-scenes/van-gogh-room.glb
Platform::GlfwApplication: warning: the executable is not a HiDPI-enabled app bundle
Illegal instruction: 4
```

Seems to be dying here: 
https://github.com/mosra/magnum/blob/b937f3901e8dcb3f108769166fcf9d3be5a43fbc/src/Magnum/Platform/GlfwApplication.cpp#L124

erikwijmans, 2019-04-23:
> Came here to report the same issue. We should really freeze the version/commit-id of magnum we use.

We have frozen the version/commit-id of magnum.  This appears to be an update to MacOS or clang that is causing this.

erikwijmans, 2019-04-23:
I have now run into this issue also (going to update Magnum, just need to fix a few compile errors).  I do not get the same issue as you @dhruvbatra, can you try a clean build (do `rm -rf build` then build).

CC: @msavva

dhruvbatra, 2019-04-23:
Clean build, seeing same error.

mosra, 2019-04-23:
> Even after ^^this fix by @titardrew, I am seeing the following error with viewer:
>
> ```
> (base) dbatra-mbp:habitat-sim dbatra$ ./build/viewer [...]

This warning is unrelated (it just checked if the app was Retina-enabled and discovered it's not, which is not a critical problem). The "Illegal instruction" seems to be a mismatch between [macOS versions it was compiled for and used on](https://stackoverflow.com/questions/14268887/what-is-the-illegal-instruction-4-error-and-why-does-mmacosx-version-min-10)? Can you get a backtrace?

msavva, 2019-05-03:
I now also encountered the above build error after a MacOS and Xcode update.  Build error is resolved using @titardrew 's fix but then I also encounter the `Illegal instruction: 4` error message when running the interactive viewer.  Backtrace below: 
```
Process:               viewer [16196]
Path:                  /Users/USER/*/viewer
Identifier:            viewer
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        zsh [43454]
Responsible:           viewer [16196]
User ID:               847061996

Date/Time:             2019-05-03 09:14:56.032 -0700
OS Version:            Mac OS X 10.14.4 (18E226)
Report Version:        12
Bridge OS Version:     3.0 (14Y677)
Anonymous UUID:        4FC35A73-E0CA-BAD4-7435-301231890324

Sleep/Wake UUID:       22F40FE1-FF31-4760-A803-DDD3C0E7A14E

Time Awake Since Boot: 210000 seconds
Time Since Wake:       8600 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [16196]

Application Specific Information:
unknown pixel format

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.AppKit              	0x00007fff34dfdcae -[NSCGLSurface flushRect:] + 4639
1   com.apple.AppKit              	0x00007fff34dfc9f4 NSCGLSurfaceFlush + 214
2   GLEngine                      	0x00007fff40965a4a glSwap_Exec + 186
3   com.apple.opengl              	0x00007fff4094774f CGLFlushDrawable + 59
4   com.apple.AppKit              	0x00007fff34756478 -[NSOpenGLContext flushBuffer] + 27
5   viewer                        	0x000000010c9052d7 _glfwRefreshContextAttribs + 1399 (context.c:614)
6   viewer                        	0x000000010c90a195 glfwCreateWindow + 501 (window.c:223)
7   viewer                        	0x000000010c7b2083 Magnum::Platform::GlfwApplication::tryCreate(Magnum::Platform::GlfwApplication::Configuration const&, Magnum::Platform::GlfwApplication::GLConfiguration const&) + 867
8   viewer                        	0x000000010c7b1b18 Magnum::Platform::GlfwApplication::tryCreate(Magnum::Platform::GlfwApplication::Configuration const&) + 72
9   viewer                        	0x000000010c7b08c0 Magnum::Platform::GlfwApplication::GlfwApplication(Magnum::Platform::GlfwApplication::Arguments const&, Magnum::Platform::GlfwApplication::Configuration const&) + 32 (GlfwApplication.cpp:106)
10  viewer                        	0x000000010c78d10e esp::gfx::Viewer::Viewer(Magnum::Platform::GlfwApplication::Arguments const&) + 270
11  viewer                        	0x000000010c7606bb main + 75 (viewer.cpp:9)
12  libdyld.dylib                 	0x00007fff6318c3d5 start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff633793f0 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff633793f0 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff633793f0 start_wqthread + 0

Thread 4:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff632c1266 semaphore_wait_trap + 10
1   libdispatch.dylib             	0x00007fff6313fbd9 _dispatch_sema4_wait + 16
2   libdispatch.dylib             	0x00007fff631403a0 _dispatch_semaphore_wait_slow + 98
3   com.apple.HIToolbox           	0x00007fff36009268 _BeginEventReceiptOnThread + 159
4   com.apple.AppKit              	0x00007fff343a1c90 _NSEventThread + 37
5   libsystem_pthread.dylib       	0x00007fff6337a2eb _pthread_body + 126
6   libsystem_pthread.dylib       	0x00007fff6337d249 _pthread_start + 66
7   libsystem_pthread.dylib       	0x00007fff6337940d thread_start + 13
[... 305 lines cut ...]
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x80AD, 0x483943434E4E4E434C47414C41522D4E5644
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x173), Broadcom BCM43xx 1.0 (7.77.61.2 AirPortDriverBrcmNIC-1305.8)
Bluetooth: Version 6.0.11f4, 3 services, 27 devices, 1 incoming serial ports
Network Service: USB 10/100/1000 LAN, Ethernet, en8
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: USB2.0 Hub
USB Device: 4-Port USB 2.0 Hub
USB Device: Hub
USB Device: Texas Instruments USBtoI2C Solution
USB Device: Hub
USB Device: Evoluent VerticalMouse 4
USB Device: Advantage2 Keyboard
USB Device: USB 2.0 BILLBOARD
USB Device: iBridge
USB Device: Yubikey 4 OTP+U2F+CCID
USB Device: USB3.0 Hub
USB Device: USB3.0 Card Reader
USB Device: 4-Port USB 3.0 Hub
USB Device: Hub
USB Device: Hub
USB Device: USB 10/100/1000 LAN
Thunderbolt Bus: MacBook Pro, Apple Inc., 39.2
Thunderbolt Bus: MacBook Pro, Apple Inc., 39.2
```

dhruvbatra, 2019-05-13:
I believe this issue should be fixed. If so, please close @titardrew

titardrew, 2019-05-14:
@dhruvbatra, sure. Thanks for this fix!

closed 2019-05-14 by titardrew