facebookresearch/habitat-sim #18: Segmentation Fault

ZwwWayne, 2019-04-16:
## Steps to reproduce
With gcc 5.4 and cmake 3.14, ld 2.32, gflags installed by anaconda, I run either build.sh --headless and python setup.py --headless. Both could compiled successfully, but could not success to run the example.

## Observed Results
I meet the segmentation fault during running the example.py with either scene.
[image]

But I do not know why.

Does anyone meet the same problem, or is there any solution to this?

erikwijmans, 2019-04-16:
Segfaults can be quite tricky to debug unfortunately.  Can you give us some more information about your system?  i.e. linux version, nvidia driver version, graphics card type, CUDA version, etc.  

Also, what does `ldd build/esp/bindings/habitat_sim_bindings.*.so` show?

Can you also make sure that no logging has been turned off, i.e. do `unset MAGNUM_LOG` and `unset GLOG_minloglevel`

ZwwWayne, 2019-04-17:
The GPU is 1080Ti, and I am using CUDA 9.0 on CentOS7.2.

Actually, `ldd build/esp/bindings/habitat_sim_bindings.*.so` shows nothing...
After compilation, the tree structure shows this 
[image]

erikwijmans, 2019-04-17:
My shot in the dark guess for what's going on is that EGL isn't getting properly picked up, given that there are no error messages.  This is unfortunately super hard to debug.  Two things to try:

1. Follow the steps for local evaluation of the challenge submission: https://github.com/facebookresearch/habitat-challenge#habitat-challenge.  This'll go through running things in a docker container and will help isolate variables.
2. Some systems need libglvnd installed for EGL to get picked up correctly, so if you don't already have it installed, I would give that a try: https://github.com/NVIDIA/libglvnd

mosra, 2019-04-23:
Not sure if related, but someone encountered a crash when using glvnd on Debian Testing few weeks ago -- on that distribution, if NVidia's glvnd gets used, context creation crashes because not even `glGetString()` is able to return correct things. But in that case the engine is able to print at least a few lines of output before blowing up.

ZwwWayne, 2019-04-23:
Ok, thanks a lot for the suggestion. 
I have already give up to install the whole environment through anaconda, and maybe I will try docker further.
Thanks

closed 2021-03-01 by dhruvbatra