facebookresearch/habitat-sim #22: [C++ -> Python] Port SemanticScene to python

erikwijmans, 2019-04-17:
The semantic scene loading code is currently written in c++.  It likely shouldn't be (I can't think of a performance reason), so we should port it to python!

dhruvbatra, 2019-04-18:
I may be mistaken, but aren't we trying to move this even lower in the stack (by asking for this to be implemented in a unified data-loader in Magnum)?

erikwijmans, 2019-04-18:
The SemanticScene contains the “metadata” for annotations. Things like the name of an object, it’s bounding box, what room it’s in, etc. I don’t believe that stuff we’d want to move that stuff down to Magnum. Also, the SemanticScene doesn’t not contain the mesh needed to do rendering for the semantic sensor.

dhruvbatra, 2019-04-18:
Makes sense. Thanks.

closed 2022-02-03 by aclegg3