facebookresearch/habitat-sim #23: example.py cannot use semantic sensor

david2611, 2019-04-18:
## Steps to reproduce

  1. Clone repo and install using ```./build.sh```
  2. ```python habitat-sim/examples/example.py --scene /path/to/habitat-test-scenes/skokloster-castle.glb --save_png --max_frames 100 --depth_sensor --semantic_sensor --print_semantic_mask_stats```

## Observed Results

```Assert failed: 0 <= activeSemanticSceneID_ && activeSemanticSceneID_ < sceneID_.size(), /path/to/habitat-sim/src/esp/gfx/Simulator.cpp:178```

## Expected Results

Expected to see saved images of semantic mask for the environment.

I am expecting that this is less of an issue with the code itself and more the test environments available as I suspect the test environment does not have semantic labels available. 

This makes the example code hard for me to use and experiment with on its own as I cannot experiment with the functionality without getting permission for, and downloading full datasets.

I am trying to determine what level of semantic mask is available to me (class-wise Vs instance-wise) and how I could utilise them in my work.

Is there a way currently to do this with just the example.py script and example data provided?

erikwijmans, 2019-04-18:
Unfortunately we have yet to find any semantically annotated meshes that we can distribute via a public URL with no license; we'd love to find one that we can distribute.

For now, I can answer questions:
1. The semantic sensor supports both instance-wise and class-wise segmentations for the Matterport3D dataset: https://github.com/niessner/Matterport.  The output of the semantic sensor is instance-wise segmentation, and then there is a mapping for object instance ID to the object's semantic name and semantic label ID.

1. Gibson doesn't have semantic/instance annotations.

david2611, 2019-04-22:
I completely understand that this would be an issue.

erikwijmans, 2019-04-23:
> Is SUNCG still supported>

Suncg is support and has instance and class semantic labels

> Is it easy to get access to 3D bounding box annotations in global coordinates for any existing environments

For Matterport3D and SUNCG, these are available and easy to access (they are just a property on each object in the semantic_scene).  Although I believe SUNCG only has axis-aligned bounding boxes.

> You say you have the ability to support generic datasets. Is there a Readme somewhere that I missed that explains how this is most easily done for 3D environments created in say Unreal

 CC @msavva 
The internal representation for a dataset is generic, but loading code for any given dataset isn't generic (we try to make it as generic as possible, but that can be challenging or impossible).  I am not sure if any of our current loaders will support unreal.

david2611, 2019-04-23:
Thanks very much for the help in answering questions. I am happy to call this issue closed.

closed 2019-04-23 by david2611