facebookresearch/habitat-sim #25: setup.py fails

danielgordon10, 2019-04-18:
## Steps to reproduce
`python setup.py install --headless`

## Observed Results
```
[ 91%] Building CXX object esp/gfx/CMakeFiles/gfx.dir/__/__/deps/magnum/src/MagnumPlugins/AnyImageImporter/importStaticPlugin.cpp.o
[ 91%] Building CXX object esp/gfx/CMakeFiles/gfx.dir/PTexMeshShader.cpp.o
In file included from /raid/xkcd/external_code/habitat-sim/src/esp/gfx/WindowlessContext.cpp:13:0:
/raid/xkcd/external_code/habitat-sim/src/esp/gfx/WindowlessContext.cpp: In constructor ‘esp::gfx::{anonymous}::ESPEGLContext::ESPEGLContext(int)’:
/raid/xkcd/external_code/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:276:43: error: expected primary-expression before ‘,’ token
 #define EGL_NO_CONTEXT EGL_CAST(EGLContext,0)
                                           ^
/raid/xkcd/external_code/habitat-sim/src/esp/gfx/WindowlessContext.cpp:151:54: note: in expansion of macro ‘EGL_NO_CONTEXT’
     context_ = eglCreateContext(display_, eglConfig, EGL_NO_CONTEXT, NULL);
                                                      ^
/raid/xkcd/external_code/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:276:45: error: ‘EGL_CAST’ was not declared in this scope
 #define EGL_NO_CONTEXT EGL_CAST(EGLContext,0)
                                             ^
/raid/xkcd/external_code/habitat-sim/src/esp/gfx/WindowlessContext.cpp:151:54: note: in expansion of macro ‘EGL_NO_CONTEXT’
     context_ = eglCreateContext(display_, eglConfig, EGL_NO_CONTEXT, NULL);
                                                      ^
/raid/xkcd/external_code/habitat-sim/src/esp/gfx/WindowlessContext.cpp: In member function ‘virtual void esp::gfx::{anonymous}::ESPEGLContext::makeCurrent()’:
/raid/xkcd/external_code/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:278:43: error: expected primary-expression before ‘,’ token
 #define EGL_NO_SURFACE EGL_CAST(EGLSurface,0)
                                           ^
/raid/xkcd/external_code/habitat-sim/src/esp/gfx/WindowlessContext.cpp:164:34: note: in expansion of macro ‘EGL_NO_SURFACE’
         eglMakeCurrent(display_, EGL_NO_SURFACE, EGL_NO_SURFACE, context_);
                                  ^
/raid/xkcd/external_code/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:278:45: error: ‘EGL_CAST’ was not declared in this scope
 #define EGL_NO_SURFACE EGL_CAST(EGLSurface,0)
                                             ^
/raid/xkcd/external_code/habitat-sim/src/esp/gfx/WindowlessContext.cpp:164:34: note: in expansion of macro ‘EGL_NO_SURFACE’
         eglMakeCurrent(display_, EGL_NO_SURFACE, EGL_NO_SURFACE, context_);
                                  ^
/raid/xkcd/external_code/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:278:43: error: expected primary-expression before ‘,’ token
 #define EGL_NO_SURFACE EGL_CAST(EGLSurface,0)
                                           ^
/raid/xkcd/external_code/habitat-sim/src/esp/gfx/WindowlessContext.cpp:164:50: note: in expansion of macro ‘EGL_NO_SURFACE’
         eglMakeCurrent(display_, EGL_NO_SURFACE, EGL_NO_SURFACE, context_);
                                                  ^
esp/gfx/CMakeFiles/gfx.dir/build.make:166: recipe for target 'esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o' failed
make[2]: *** [esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:4465: recipe for target 'esp/gfx/CMakeFiles/gfx.dir/all' failed
make[1]: *** [esp/gfx/CMakeFiles/gfx.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
Traceback (most recent call last):
  File "setup.py", line 147, in <module>
    zip_safe=False,
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 43, in run
    self.build_extension(ext)
  File "setup.py", line 100, in build_extension
    ["cmake", "--build", "."] + build_args, cwd=self.build_temp
  File "/raid/xkcd/miniconda3/envs/habitat-env/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.
```

danielgordon10, 2019-04-18:
For the record, build.sh seems to be working fine.

danielgordon10, 2019-04-19:
This may be somehow caused by using ninja instead of cmake. Not entirely sure.

erikwijmans, 2019-04-19:
Can you provide more information about your system? 

OS, version, g++ version, etc

erikwijmans, 2019-04-19:
Actually, if I remember correctly, this is from having an old libegl1-mesa-dev installed. Can you update that package? I can also get you the header file from that if needed.

danielgordon10, 2019-04-19:
How do I update that? I think it was installed by the nvidia drivers installation which I don't want to mess with.

danielgordon10, 2019-04-19:
Huzzah.
`sudo apt install libegl1-mesa-dev` did the trick.

closed 2019-04-19 by danielgordon10