facebookresearch/habitat-sim #26: Cannot import habitat_sim anymore

danielgordon10, 2019-04-18:
https://github.com/facebookresearch/habitat-sim/blob/8ee3eb45d44cbcd51850d977136fbbe5f9dedaa7/habitat_sim/agent/agent.py#L7

```
>>> import habitat_sim
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/raid/xkcd/external_code/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .agent import *
  File "/raid/xkcd/external_code/habitat-sim/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/raid/xkcd/external_code/habitat-sim/habitat_sim/agent/agent.py", line 7, in <module>
    import attr
ModuleNotFoundError: No module named 'attr'
```

I tried pip install attr but then I get
```
Type "help", "copyright", "credits" or "license" for more information.
>>> import habitat_sim
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/raid/xkcd/external_code/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .agent import *
  File "/raid/xkcd/external_code/habitat-sim/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/raid/xkcd/external_code/habitat-sim/habitat_sim/agent/agent.py", line 9, in <module>
    from .controls import ObjectControls, ActuationSpec
  File "/raid/xkcd/external_code/habitat-sim/habitat_sim/agent/controls.py", line 17, in <module>
    @attr.s(auto_attribs=True, slots=True)
AttributeError: module 'attr' has no attribute 's'
```
So I don't know what attr package is.

erikwijmans, 2019-04-19:
It’s in requirements.txt. It’s called attrs on pip, so pip install attrs will do it.

danielgordon10, 2019-04-19:
Whoops, that one's on me.

closed 2019-04-19 by danielgordon10