facebookresearch/habitat-sim #28: Error in `python': munmap_chunk(): invalid pointer: 0x00007f48777bb770

danielgordon10, 2019-04-19:
I don't really know what's causing this. I have built from `HEAD` and get this issue, but building from `99e60b97bb736218799424645c22d0258e01c0a7` does not. The issue doesn't happen when running example.py, but that's the only thing I really know for sure. The failure happens when calling vector_env reset.
```
Restored logs/pointnav/mp3d/splitnet_rl/checkpoints/2019_04_10_17_27_49/000004912.pt
2019-04-19 13:17:12,504 initializing sim Sim-v0
Renderer: GeForce GTX TITAN X/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 410.72
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_EXT_direct_state_access
    GL_KHR_debug
Using driver workarounds:
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
2019-04-19 13:17:13,812 initializing task Nav-v0
Envs created in 1.309
Writing results to logs/pointnav/mp3d/splitnet_rl/results/val/logs_pointnav_mp3d_splitnet_rl_2019_04_10_17_27_49_000004912.csv
Renderer: GeForce GTX TITAN X/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 410.72
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_EXT_direct_state_access
    GL_KHR_debug
Using driver workarounds:
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
*** Error in `python': munmap_chunk(): invalid pointer: 0x00007f48777bb770 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f4932c537e5]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x1a8)[0x7f4932c60698]
/home/xkcd/Documents/external_code/habitat-sim/build/esp/bindings/habitat_sim_bindings.cpython-36m-x86_64-linux-gnu.so(+0xcb5e1)[0x7f48e72485e1]
/home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/torch/lib/libtorch_python.so(+0x1105fa)[0x7f49238625fa]
/home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/torch/lib/libtorch_python.so(+0x11085e)[0x7f492386285e]
python(+0x1984a8)[0x56542bab94a8]
python(_PyEval_EvalFrameDefault+0x30a)[0x56542bade38a]
python(+0x1918e4)[0x56542bab28e4]
python(+0x192771)[0x56542bab3771]
python(+0x198505)[0x56542bab9505]
python(_PyEval_EvalFrameDefault+0x30a)[0x56542bade38a]
python(+0x191bfe)[0x56542bab2bfe]
python(+0x192771)[0x56542bab3771]
python(+0x198505)[0x56542bab9505]
python(_PyEval_EvalFrameDefault+0x30a)[0x56542bade38a]
[... 265 lines cut ...]
7f48c32e7000-7f48c32e9000 r--p 00000000 09:00 20608242                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/numba/_dynfunc.cpython-36m-x86_64-linux-gnu.so
7f48c32e9000-7f48c32ea000 r-xp 00002000 09:00 20608242                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/numba/_dynfunc.cpython-36m-x86_64-linux-gnu.so
  48c32ea000-7f48c32eb000 r--p 00003000 09:00 20608242                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/numba/_dynfunc.cpython-36m-x86_64-linux-gnu.so
7f48c32eb000-7f48c32ec000 r--p 00003000 09:00 20608242                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/numba/_dynfunc.cpython-36m-x86_64-linux-gnu.so
7f48c32ec000-7f48c32ed000 rw-p 00004000 09:00 20608242                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/numba/_dynfunc.cpython-36m-x86_64-linux-gnu.so
7f48c32ed000-7f48c332d000 rw-p 00000000 00:00 0 
7f48c332d000-7f48c332f000 r--p 00000000 09:00 20608236                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/numba/typeconv/_typeconv.cpython-36m-x86_64-linux-gnu.so
7f48c332f000-7f48c3331000 r-xp 00002000 09:00 20608236                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/numba/typeconv/_typeconv.cpython-36m-x86_64-linux-gnu.so
7f48c3331000-7f48c3332000 r--p 00004000 09:00 20608236                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/numba/typeconv/_typeconv.cpython-36m-x86_64-linux-gnu.so
7f48c3332000-7f48c3333000 r--p 00004000 09:00 20608236                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/numba/typeconv/_typeconv.cpython-36m-x86_64-linux-gnu.so
7f48c3333000-7f48c3334000 rw-p 00005000 09:00 20608236                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/numba/typeconv/_typeconv.cpython-36m-x86_64-linux-gnu.so
7f48c3334000-7f48c3534000 rw-p 00000000 00:00 0 
7f48c3534000-7f48c3691000 r--p 00000000 09:00 20646207                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/llvmlite/binding/libllvmlite.so
7f48c3691000-7f48c4bc9000 r-xp 0015d000 09:00 20646207                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/llvmlite/binding/libllvmlite.so
7f48c4bc9000-7f48c5891000 r--p 01695000 09:00 20646207                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/llvmlite/binding/libllvmlite.so
7f48c5891000-7f48c5892000 ---p 0235d000 09:00 20646207                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/llvmlite/binding/libllvmlite.so
7f48c5892000-7f48c5a03000 r--p 0235d000 09:00 20646207                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/llvmlite/binding/libllvmlite.so
7f48c5a03000-7f48c5a0c000 rw-p 024ce000 09:00 20646207                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/llvmlite/binding/libllvmlite.so
7f48c5a0c000-7f48c5ac0000 rw-p 00000000 00:00 0 
7f48c5ac0000-7f48c5ac3000 r--p 00000000 09:00 20608414                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/quaternion/numpy_quaternion.cpython-36m-x86_64-linux-gnu.so
7f48c5ac3000-7f48c5ad5000 r-xp 00003000 09:00 20608414                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/quaternion/numpy_quaternion.cpython-36m-x86_64-linux-gnu.so
7f48c5ad5000-7f48c5ada000 r--p 00015000 09:00 20608414                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/quaternion/numpy_quaternion.cpython-36m-x86_64-linux-gnu.so
7f48c5ada000-7f48c5adb000 r--p 00019000 09:00 20608414                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/quaternion/numpy_quaternion.cpython-36m-x86_64-linux-gnu.so
7f48c5adb000-7f48c5adc000 rw-p 0001a000 09:00 20608414                   /home/xkcd/miniconda3/envs/habitat-env/lib/python3.6/site-packages/quaternion/numpy_quaternion.cpython-36m-x86_64-linux-gnu.soAborted (core dumped)
```

danielgordon10, 2019-04-19:
The issue is caused by https://github.com/facebookresearch/habitat-sim/commit/4206144d2e86e8af59dadff8e362fb9371e8c3dc but I don't know what in it.

danielgordon10, 2019-04-19:
I can't 100% verify, but @erikwijmans seems confident that 8ee3eb45d44cbcd51850d977136fbbe5f9dedaa7 will fix the issue. I can't pull from master because it isn't backwards compatible with the habitat-api I'm using.

danielgordon10, 2019-04-19:
I have verified the fix works now thanks to lots of help from @erikwijmans

closed 2019-04-19 by danielgordon10

erikwijmans, 2019-04-19:
If anyone else runs into this but can't be on current master, you can cherry-pick the fix via `git cherry-pick 8ee3eb45d44cbcd51850d977136fbbe5f9dedaa7`