facebookresearch/habitat-sim #32: Interactive viewer does not work

takuma-ynd, 2019-04-23:
## Steps to reproduce
I am on MacOS with XQuartz 2.7.11 (xorg-server 1.18.4)

  1. Build docker image based on [habitat-api repository's Dockerfile](https://github.com/facebookresearch/habitat-api/blob/master/Dockerfile) without `--headless` option at [line 45](https://github.com/facebookresearch/habitat-api/blob/8c5329ce9f777395b45c2daac135377e629ec148/Dockerfile#L45)
  2. run container with X11 forwarding `nvidia-docker run -it --rm --net host --env "DISPLAY" --volume="$HOME/.Xauthority:/root/.Xauthority:rw" -v /tmp/.X11-unix:/tmp/.X11-unix habitat-sim bash`
  3. run `build/viewer [some scene data]`

## Observed Results
I got the following error:
```
$ ./build/viewer 
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
Invalid window size -2147483648x-2147483648
Platform::GlfwApplication::tryCreate(): cannot create a window with core OpenGL context, falling back to compatibility context
Invalid window size -2147483648x-2147483648
Platform::GlfwApplication::tryCreate(): cannot create a window with OpenGL context
```

## Expected Results
The viewer should appear.

## Other observations
I confirmed that some X apps like `xeyes`, `xclock` work as expected.
In addition, `glxgears` also works as usual.

StOnEGiggity, 2019-05-28:
Hi, I meet the similar problem on Ubuntu 16.04.6 with two GeForce GTX 1080 Ti and one Tesla K40c.
1. I log in the server via ssh -X -Y name@ip and I am on MacOS with XQuartz 2.7.8.
2. I follow the quick installation and everything is ok.
3. I run the interactive testing code but find the same error as above.
4. I follow the guide from #1 , same error occurs and it is ok when I add `--headless`

I guess some dependency vision may be wrong. Is that possible to provide some specific version for the library?

mathfac, 2019-06-22:
@takuma-ynd, @StOnEGiggity  thank you for reporting the issue. Is still doesn't work on latest commit of habitat-sim?

anoopsonar30, 2019-07-02:
@mathfac I'm facing the same issue on the latest commit of habitat-sim as well.

anoopsonar30, 2019-07-03:
I seem to be getting the following error now. I'll try installing OpenGL version 2.1 and see if that helps. 

> (habitat) root@ip-172-31-65-173:/habitat-sim# build/viewer ../habitat-api/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb 
> debug1: client_input_channel_open: ctype x11 rchan 3 win 65536 max 16384
> debug1: client_request_x11: request from 127.0.0.1 49476
> debug1: channel 1: new [x11]
> debug1: confirm x11
> debug1: client_input_channel_open: ctype x11 rchan 4 win 65536 max 16384
> debug1: client_request_x11: request from 127.0.0.1 49478
> debug1: channel 2: new [x11]
> debug1: confirm x11
> Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
> debug1: channel 2: FORCE input drain
> debug1: channel 2: free: x11, nchannels 3
> GLX: Forward compatibility requested but GLX_ARB_create_context_profile is unavailable
> Platform::GlfwApplication::tryCreate(): cannot create a window with core OpenGL context, falling back to compatibility context
> Requested OpenGL version 2.1, got version 1.4
> Platform::GlfwApplication::tryCreate(): cannot create a window with OpenGL context
> debug1: channel 1: FORCE input drain

anoopsonar30, 2019-07-04:
@mathfac @erikwijmans Is it possible this might be due to inconsistent OpenGL versions?
Running glxinfo gives this - 

> (habitat) ubuntu@ip-172-31-46-89:~/habitat-source/sim-vis$ glxinfo | grep -i opengdebug1: client_input_channel_open: ctype x11 rchan 3 win 65536 max 16384
> debug1: client_request_x11: request from 127.0.0.1 58208
> debug1: channel 1: new [x11]
> debug1: confirm x11
> l
> OpenGL vendor string: Intel Inc.
> OpenGL renderer string: Intel HD Graphics 5000 OpenGL Engine
> OpenGL version string: 1.4 (2.1 INTEL-10.6.33)
> OpenGL extensions:
> debug1: channel 1: FORCE input drain

which clearly doesn't use nvidia's OpenGL 4.6 driver which was installed with the nvidia-430 graphics driver. How do I make it use that? The nvidia driver is used correctly when I run examples/benchmark.py on the scene files.

erikwijmans, 2019-07-04:
X11 forwarding is super strange -- It forwards the OpenGL calls to your _local_ machine and then uses your _local_ machine's resources to render things.

KenaHemnani, 2020-10-18:
Using the test Interactive viewer I am getting the following error:
**Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to ph                 ysical DPI
GLX: Failed to create context: GLXBadFBConfig
Platform::GlfwApplication::tryCreate(): cannot create a window with core OpenGL                  context, falling back to compatibility context
GLX: Failed to create context: GLXBadFBConfig
Platform::GlfwApplication::tryCreate(): cannot create a window with OpenGL conte                 xt
**
I have connected to remote server via. MobaXterm and have done installation on a virtual machine.
Can someone please suggest how to resolve?

PatZhuang, 2021-01-06:
Running on NVIDIA docker with vnc4server, I got similar issue when running `./build/viewer`

```
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
GLX: GLX version 1.3 is required
Platform::GlfwApplication::tryCreate(): cannot create a window with core OpenGL context, falling back to compatibility context
GLX: No GLXFBConfigs returned
GLX: Failed to find a suitable GLXFBConfig
Platform::GlfwApplication::tryCreate(): cannot create a window with OpenGL context
```

The `glxinfo` shows:

```
name of display: :1
display: :1  screen: 0
direct rendering: No (If you want to find out why, try setting LIBGL_DEBUG=verbose)
server glx vendor string: SGI
server glx version string: 1.2
server glx extensions:
    GLX_EXT_import_context, GLX_EXT_visual_info, GLX_EXT_visual_rating
client glx vendor string: NVIDIA Corporation
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_stereo_tree, GLX_EXT_swap_control, 
    GLX_EXT_swap_control_tear, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_NV_copy_buffer, 
    GLX_NV_copy_image, GLX_NV_delay_before_swap, GLX_NV_float_buffer, 
    GLX_NV_multisample_coverage, GLX_NV_present_video, 
    GLX_NV_robustness_video_memory_purge, GLX_NV_swap_group, 
    GLX_NV_video_capture, GLX_NV_video_out, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.2
GLX extensions:
    GLX_ARB_get_proc_address, GLX_EXT_import_context, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating
OpenGL vendor string: Mesa project: www.mesa3d.org
OpenGL renderer string: Mesa GLX Indirect
OpenGL version string: 1.3 Mesa 4.0.4
OpenGL extensions:
    GL_ARB_imaging, GL_ARB_multitexture, GL_ARB_texture_border_clamp, 
    GL_ARB_texture_cube_map, GL_ARB_texture_env_add, 
    GL_ARB_texture_env_combine, GL_ARB_texture_env_dot3, 
    GL_ARB_transpose_matrix, GL_EXT_abgr, GL_EXT_blend_color, 
    GL_EXT_blend_minmax, GL_EXT_blend_subtract, GL_EXT_texture_env_add, 
    GL_EXT_texture_env_combine, GL_EXT_texture_env_dot3, 
    GL_EXT_texture_lod_bias

7 GLX Visuals
    visual  x   bf lv rg d st  colorbuffer  sr ax dp st accumbuffer  ms  cav
  id dep cl sp  sz l  ci b ro  r  g  b  a F gb bf th cl  r  g  b  a ns b eat
----------------------------------------------------------------------------
0x025 16 tc  1  16  0 r  y .   6  5  5  0 .  .  0 16  0  0  0  0  0  0 0 None
0x026 16 tc  1  16  0 r  y .   6  5  5  0 .  .  0 16  8 16 16 16  0  0 0 None
0x027 16 tc  1  16  0 r  y .   6  5  5  8 .  .  0 16  8 16 16 16 16  0 0 None
0x028 16 dc  1  16  0 r  y .   6  5  5  0 .  .  0 16  0  0  0  0  0  0 0 None
0x029 16 dc  1  16  0 r  y .   6  5  5  0 .  .  0 16  8 16 16 16  0  0 0 None
0x02a 16 dc  1  16  0 r  y .   6  5  5  8 .  .  0 16  8 16 16 16 16  0 0 None
0x02b 16 dc  1  16  0 r  . .   6  5  5  8 .  .  0 16  8 16 16 16 16  0 0 None
```