facebookresearch/habitat-sim #34: [Question] Modify movement step-size in interactive viewer

brucecui97, 2019-04-26:
May I please ask if it is possible to modify the step-size of say forward movement or rotation to be finer (e.g. 0.05 units) in the interactive viewer? Thank you for your help in advance.

erikwijmans, 2019-04-26:
We currently don't have those exposed as command-line options, but they can be changed in the source and recompile: https://github.com/facebookresearch/habitat-sim/blob/master/src/esp/gfx/Viewer.cpp#L19

brucecui97, 2019-04-26:
Thanks. As a related question, may I please ask if it is possible to use the interactive viewer to view what the robot is doing while running example.py?

erikwijmans, 2019-04-26:
It isn't, but there are two options for viewing what an agent is doing:

1. `example.py` has a `--save_png` option that'll dump out the options as png images to disk.
1. This PR over on habitat-api: https://github.com/facebookresearch/habitat-api/pull/76 introduces an example of how to make a video showing observations and the agent's path on a topdown map.

closed 2019-04-26 by brucecui97