facebookresearch/habitat-sim #36: Passing commands to pip install

danielgordon10, 2019-05-07:
It looks like the current `setup.py` script does not support the `--install-option` arg.
https://stackoverflow.com/questions/18725137/how-to-obtain-arguments-passed-to-setup-py-from-pip-with-install-option

mosra, 2019-05-07:
Nice, just today I was wondering if there's a canonical way to pass things through to the underlying cmake invocation. This could be it.

erikwijmans, 2019-05-08:
I've looked as this before.  Unfortunately install options don't work for both `install` and `build_ext`.   Which would limit us to just using `install` and `develop` for development workflow, but neither are particularly nice when you have both python and c++ code. The pattern I see from other repos that have both python and c++ code is to parse the args yourself in `setup.py` and then use environment variables to interact with pip.  We can add a `--cmake-args` command line arg and a `CMAKE_ARGS` environment variable

mosra, 2019-05-09:
I ended up configuring the cmake build manually and then redirecting `setup.py` to use the existing directory instead of creating its own, for example:

```sh
python setup.py install build_ext --build-temp=build
```

For my needs this has all the flexibility I want, however might not be convenient enough for others.

mathfac, 2019-06-22:
Should be solved.

closed 2019-06-22 by mathfac