facebookresearch/habitat-sim #42: Update to use modern CMake practices

mosra, 2019-05-10:
I'll tackle this once I'm done with current priority tasks. Things to do:

- [ ] replace `include_directories()` with `target_include_directories()`, preferring `PRIVATE` where possible; if a dependency provides a CMake target use that instead
- [ ] remove use of `file(GLOB)`, list sources explicitly (makes the build more predictable without things getting broken when there are untracked files in the working copy)
- [ ] convert `if(${FOO})` to just `if(FOO)`
- [ ] look into TODOs with [CMAKE_CXX_VISIBILITY_PRESET](https://github.com/facebookresearch/habitat-sim/blob/e7d115f034020ae7b91b2d1566ec787fe188580f/src/CMakeLists.txt#L37)

Skylion007, 2020-10-13:
@mosra Can this be closed?

mosra, 2020-10-14:
Why? There's still a few good-to-have items from the list that are not done yet. Feel free to tackle those, if you have time.

aclegg3, 2022-08-31:
@mosra any more updates here?

mosra, 2022-08-31:
Hmm. Do we have any other place to list such low-priority tech debt TODOs? In my case (with Magnum repos) I have a bunch of such long-lived "aggregate issues" full of checkboxes that act more as a reminder of what needs to be *eventually* solved, and I opened this one with a similar intent. 

But if the aim here is to only have an issue open if it's likely to be resolved soon, feel free to close.

aclegg3, 2022-08-31:
I'll leave this open for now until we establish such a list and then we can migrate the remainder there.