facebookresearch/habitat-sim #47: setup.py fails after installing libegl1-mesa-dev

medhagitnod, 2019-05-20:
## Steps to reproduce

  1. sudo apt install libegl1-mesa-dev
  2. python setup.py install
Using Ubuntu 18.04

## Observed Results
[ 24%] Linking CXX static library libtinyply.a
[ 24%] Built target tinyply
make: *** [Makefile:152: all] Error 2
Traceback (most recent call last):
  File "setup.py", line 323, in <module>
    zip_safe=False,
  File "/home/medha/habitat2/lib/python3.7/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/medha/habitat2/lib/python3.7/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/home/medha/habitat2/lib/python3.7/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/medha/habitat2/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/medha/habitat2/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/medha/habitat2/lib/python3.7/site-packages/setuptools/command/install_lib.py", line 24, in run
    self.build()
  File "/usr/lib/python3.7/distutils/command/install_lib.py", line 109, in build
    self.run_command('build_ext')
  File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 166, in run
    self.build_extension(ext)
  File "setup.py", line 220, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.

erikwijmans, 2019-05-21:
Is there any output from g++?

medhagitnod, 2019-05-21:
sudo apt-get install libxi-dev did the trick

closed 2019-05-21 by medhagitnod