facebookresearch/habitat-sim #48: ModuleNotFoundError: No module named 'habitat_sim._ext'

medhagitnod, 2019-05-21:
## Steps to reproduce

  1. ./build.sh
  2. Download habitat-test-scenes from provided link
  3. python examples/example.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb

## Observed Results

  Traceback (most recent call last):
  File "examples/example.py", line 12, in <module>
    import demo_runner as dr
  File "/home/medha/python-virtual-environments/habitat/habitat-sim/examples/demo_runner.py", line 15, in <module>
    import habitat_sim
  File "/home/medha/python-virtual-environments/habitat/lib/python3.7/site-packages/habitat_sim-0.0.1-py3.7-linux-x86_64.egg/habitat_sim/__init__.py", line 12, in <module>
    from .agent import *
  File "/home/medha/python-virtual-environments/habitat/lib/python3.7/site-packages/habitat_sim-0.0.1-py3.7-linux-x86_64.egg/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/home/medha/python-virtual-environments/habitat/lib/python3.7/site-packages/habitat_sim-0.0.1-py3.7-linux-x86_64.egg/habitat_sim/agent/agent.py", line 12, in <module>
    import habitat_sim.bindings as hsim
  File "/home/medha/python-virtual-environments/habitat/lib/python3.7/site-packages/habitat_sim-0.0.1-py3.7-linux-x86_64.egg/habitat_sim/bindings/__init__.py", line 25, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ModuleNotFoundError: No module named 'habitat_sim._ext'

For reference the python path is: PYTHONPATH:/path/to/habitat-sim

erikwijmans, 2019-05-21:
Looks like you have the package installed via `python setup.py install`, the code is being picked up from `/home/medha/python-virtual-environments/habitat/lib/python3.7/site-packages/habitat_sim-0.0.1-py3.7-linux-x86_64.egg`.  Can you either remove it from your site-packages or build with `python setup.py install`?

medhagitnod, 2019-05-21:
I tried to build with python.setup.py install and running the same line for the non-interactive example yields: 
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0521 13:17:58.356657 19072 ResourceManager.cpp:42] Cannot load from file /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
E0521 13:17:58.356675 19072 Simulator.cpp:84] cannot load /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
I0521 13:17:58.356726 19072 SceneManager.h:24] Deconstructing SceneManager
I0521 13:17:58.356731 19072 SceneGraph.h:20] Deconstructing SceneGraph
I0521 13:17:58.356735 19072 ResourceManager.h:46] Deconstructing ResourceManager
I0521 13:17:58.356739 19072 Renderer.cpp:37] Deconstructing Renderer
I0521 13:17:58.356935 19072 WindowlessContext.cpp:236] Deconstructing GL context
Traceback (most recent call last):
  File "examples/example.py", line 61, in <module>
    perf = demo_runner.example()
  File "/home/medha/python-virtual-environments/habitat/habitat-sim/examples/demo_runner.py", line 244, in example
    start_state = self.init_common()
  File "/home/medha/python-virtual-environments/habitat/habitat-sim/examples/demo_runner.py", line 184, in init_common
    self._sim = habitat_sim.Simulator(self._cfg)
  File "<attrs generated init b29e31f048c54c32e09b8066019eb40f5fdb3502>", line 6, in __init__
  File "/home/medha/python-virtual-environments/habitat/lib/python3.7/site-packages/habitat_sim-0.0.1-py3.7-linux-x86_64.egg/habitat_sim/simulator.py", line 34, in __attrs_post_init__
    self.reconfigure(config)
  File "/home/medha/python-virtual-environments/habitat/lib/python3.7/site-packages/habitat_sim-0.0.1-py3.7-linux-x86_64.egg/habitat_sim/simulator.py", line 82, in reconfigure
    self._config_backend(config)
  File "/home/medha/python-virtual-environments/habitat/lib/python3.7/site-packages/habitat_sim-0.0.1-py3.7-linux-x86_64.egg/habitat_sim/simulator.py", line 55, in _config_backend
    self._sim = hsim.SimulatorBackend(config.sim_cfg)
ValueError: Cannot load: /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb

msavva, 2019-05-21:
@medhagitnod thanks for trying out the Habitat platform.  The example command `python examples/example.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb` assumes you downloaded the linked test scene datasets at the quoted path.  You can either download and extract to a path of your choosing (and use that path as the directory in the argument to `--scene`) or invoke the `examples/example.py` script without an argument, in which case it will automatically download and extract the test scenes.

closed 2019-05-28 by medhagitnod