facebookresearch/habitat-sim #50: SUNCG dataset test version

edbeeching, 2019-05-24:
Hello, you mention in the readme to contact you for the SUNCG test dataset.

Also, I have the full SUNCG data, is there some sort of parsing tool/script to convert the dataset to glb files?

edbeeching, 2019-05-29:
Regarding this, I see that this [pull request](https://github.com/facebookresearch/habitat-sim/pull/55) will remove the text referring the the SUNCG dataset. Do you plan to not support the use of this dataset in the Habitat simulator? Should I revert to the House3D simulator?

Thanks.

dhruvbatra, 2019-05-29:
SUNCG continues to be supported by our code. We removed that text to avoid any confusion -- that SUNCG data needs to be requested from the original source and not from us.

erikwijmans, 2019-05-29:
SUNCG can be loaded by pointing the simulator to the `house.json` file, although I believe the `.obj` files will need to be converted to `.glb`.  You can follow along with #54 or wait for #57 to see if that resolves it :)

closed 2021-03-01 by dhruvbatra