facebookresearch/habitat-sim #51: Rename LookLeft and LookRight to RotateLeft and RotateRight

mathfac, 2019-05-24:
author @danielgordon10:
https://github.com/fairinternal/habitat-sim/blob/439d586b8c643fd42f53a31a4b15a65af3e39bcf/src/esp/scene/ObjectControls.cpp#L46

Rotate is a better description of what's actually happening since the entire agent body is being rotated along with the camera.
Based on my understanding of previous conversations, we would __not__ want to change lookUp and lookDown to rotate.

This change will have to be synced with one on habitat-api to similarly rename things.

From @dhruvbatra 

> -- "look right/left" --> "turn right/left"? Because after "looking" semantics of "forward" don't necessarily change, but after "turning" they do. 

@erikwijmans:
turnRight/turnLeft already sneakily exist: https://github.com/fairinternal/habitat-sim/blob/master/src/esp/scene/ObjectControls.cpp#L84.

lookRight/lookLeft just need to be removed from the list of body actions and then those will no longer alter the semantics of forward: https://github.com/fairinternal/habitat-sim/blob/master/src/esp/agent/Agent.cpp#L19

erikwijmans, 2019-05-24:
I added `turn_left` and `turn_right` when refactoring controls: https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/agent/default_controls.py#L75-L76

erikwijmans, 2019-05-24:
habitat-api also uses those for the turn actions: https://github.com/facebookresearch/habitat-api/blob/master/habitat/sims/habitat_simulator.py#L200

mathfac, 2019-05-29:
@erikwijmans, then we just need to clean up and  rename functions with `look right/left` names?

erikwijmans, 2019-05-29:
The look actions still have reason to exist as actions that solely move the camera (this is their functionality now).

dhruvbatra, 2019-05-29:
Agreed, we should leave the "look" actions, as their use/semantics are distinctly useful from the "turn" actions.

mathfac, 2019-05-29:
> The look actions still have reason to exist as actions that solely move the camera (this is their functionality now).

Not sure I fully understand the difference between `turn_left` and `look_left` here.
Based on [this lines of code](https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/agent/default_controls.py#L75-L76):
```
register_move_fn(look_left, "turn_left")
register_move_fn(look_right, "turn_right")
```
actions look equal.

erikwijmans, 2019-05-29:
The `turn_*`  variants are body-actions: https://github.com/facebookresearch/habitat-sim/blob/master/habitat_sim/agent/agent.py#L22 so they will turn the body of the agent.

closed 2019-05-30 by erikwijmans