facebookresearch/habitat-sim #54: Importing a mesh into Habitat

medhagitnod, 2019-05-28:
Is it possible to import or create a 3D scene in Habitat from an .obj and .mtl file of the space so that the mesh can be explored using Habitat-sim agents?

msavva, 2019-05-28:
You can use tools such as [assimp](https://github.com/assimp/assimp) to package your obj+mtl format mesh to glTF v2 binary format which can be used with Habitat.  A rough outline of steps for a given scene mesh `<scene>.obj`:
1. `brew install assimp` on MacOS, or `apt-get install assimp` etc. in Linux
1. `assimp export <scene>.obj <scene>.glb -embtex`
1. Compile Habitat and use data tool to precompute navmesh for scene: `build/utils/datatool/datatool create_navmesh <scene>.glb <scene>.navmesh`
1. Run `examples/example.py --scene <scene>.glb` or `build/viewer <scene>.glb` (NOTE: this assumes the `<scene>.navmesh` file is in the same directory as `<scene>.glb`

You can also check that the glb mesh is packaged correctly by loading it into a viewer such as https://gltf-viewer.donmccurdy.com .  Hope the above information is helpful.  Feel free to follow up here if you encounter any trouble, or just let us know if all works as expected.

medhagitnod, 2019-05-28:
When I attempt to load the glb mesh into the viewer linked, I receive the following message: THREE.GLTFLoader: Legacy binary file detected. Use LegacyGLTFLoader instead. 
Will this affect my ability to load the mesh into Habitat? If so, what can I do to ensure the glb is packaged correctly?

msavva, 2019-05-29:
The output might have been exported as a glTF v1 format as opposed to v2 format.  You can force v2 by calling `assimp export <scene>.obj <scene>.glb -fglb2 -embtex` instead.

edbeeching, 2019-05-29:
I have had similar issue with conversion with the assimp package. I found the [obj2gltf](https://github.com/AnalyticalGraphicsInc/obj2gltf) package to work out of the box. 

usage: obj2gltf -i house.obj -o house.glb

mosra, 2019-05-29:
Frankly I think this is making everyone's life unnecessarily hard for no reason :) Magnum has [AnySceneImporter](https://doc.magnum.graphics/magnum/classMagnum_1_1Trade_1_1AnySceneImporter.html) that's able to delegate to concrete importer plugins based on detected file type, so we don't need to [hardcode just glTF](https://github.com/facebookresearch/habitat-sim/blob/fa94fbf51e7c1f13933afa92d324f69e9127bd23/src/esp/assets/ResourceManager.cpp#L199-L208). And since the code is already using Assimp elsewhere, enabling [AssimpImporter](https://doc.magnum.graphics/magnum/classMagnum_1_1Trade_1_1AssimpImporter.html) for arbitrary other formats is quite trivial. 

PR submitted: #57

dhruvbatra, 2019-05-29:
Thanks @mosra!

erikwijmans, 2019-05-29:
One thing to keep in mind when loading other meshes:  The simulator, including things like that datatool that generates navmeshes, assumes -Y is the gravity direction, so you may need to rotate a given mesh to correspond with that.

medhagitnod, 2019-07-03:
the obj2gltf package did the trick

medhagitnod, 2019-07-15:
When importing the converted .obj file into Habitat-SIM there is an issue where the textures that are supposed to be on the inside of the cubical mesh have been rendered on the outside of the mesh. Does Habitat-SIM support import of the .obj / .mtl files themselves yet because the obj2gltf package creates an unusable .glb file?

mosra, 2019-07-18:
> Does Habitat-SIM support import of the .obj / .mtl files themselves yet

@medhagitnod yes, #57 adding support for this has been merged over a month ago, try it out :)

mathfac, 2019-09-05:
Closing because of inactivity, added "FAQ" label as the question is useful for others.

closed 2019-09-05 by mathfac

Rendok, 2023-08-09:
Sorry for reopening an old question. I'm following the step but on the `build/utils/datatool/Datatool create_navmesh <scene>.glb <scene>.navmesh`
I'm getting an error `ESP_CHECK failed: esp::logging::LoggingContext: No current logging context.`