facebookresearch/habitat-sim #78: non-interactive testing by running the example script does not work

weilinie, 2019-06-14:
## Steps to reproduce

I am using MacOS Mojave 10.14.5 (18F132) and running the following:
`python examples/example.py --scene `/path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`

## Observed Results

> ImportError: dlopen(/usr/local/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-macosx-10.13-x86_64.egg/habitat_sim/_ext/habitat_sim_bindings.so, 2): Symbol not found: __Py_ZeroStruct
> Referenced from: /usr/local/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-macosx-10.13-x86_64.egg/habitat_sim/_ext/habitat_sim_bindings.so
> Expected in: flat namespace
>  in /usr/local/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-macosx-10.13-x86_64.egg/habitat_sim/_ext/habitat_sim_bindings.so

## Other Observations
The interactive testing is working well.

erikwijmans, 2019-06-15:
This seems to be a mismatch between the python version targeted for complication vs. the python version used.  I'm not entirely certain how that happened as setup.py _should_ make sure those are the same.  The recommended way to install habitat-sim is in a conda environment (https://docs.conda.io/en/latest/miniconda.html), can you try that?

weilinie, 2019-06-19:
Thanks for your reply!

closed 2019-06-19 by weilinie