facebookresearch/habitat-sim #79: Pytorch integration ?

phongnhhn92, 2019-06-18:
Hello,
I have been reading Habitat AI paper and I have a question related to Pytorch integration which has been mentioned in the paper :"In practice, we have observed that it is often faster to generate images using Habitat-Sim than to load images from disk".
May I ask where can I find an example to capture RGB and depth images directly from Habitat and use them to train my own CNN model. I have tried to read the doc but I can noyt find it anywhere.
Sorry for not following the question format since I guess this is not a technical issue about Habitat. Thanks !

erikwijmans, 2019-06-18:
We don’t have an example of that, but a project using habitat has done that themselves: https://github.com/facebookresearch/splitnet/blob/master/supervised_learning/splitnet_pretrain.py

phongnhhn92, 2019-06-18:
Great, I will look into that.

closed 2019-06-18 by phongnhhn92