facebookresearch/habitat-sim #80: How to get camera intrinsic matrix?

placeforyiming, 2019-06-19:
Is there any way to get intrinsic matrix from habitat-sim?

erikwijmans, 2019-06-19:
It's a simple pinhole-camera model.  I'd recommend taking a look at this notebook: https://github.com/facebookresearch/habitat-api/blob/master/notebooks/relative_camera_views_transform_and_warping_demo.ipynb

placeforyiming, 2019-06-19:
> It's a simple pinhole-camera model. I'd recommend taking a look at this notebook: https://github.com/facebookresearch/habitat-api/blob/master/notebooks/relative_camera_views_transform_and_warping_demo.ipynb

Very useful, will check soon, thanks!

placeforyiming, 2019-06-19:
> It's a simple pinhole-camera model. I'd recommend taking a look at this notebook: https://github.com/facebookresearch/habitat-api/blob/master/notebooks/relative_camera_views_transform_and_warping_demo.ipynb

Could you tell me  which checkout is the first one to accommodate  jupyter notebook?

I meet those two issues 

[](https://github.com/facebookresearch/habitat-api/issues/134)

[](https://github.com/facebookresearch/habitat-sim/issues/81)

placeforyiming, 2019-06-20:
> It's a simple pinhole-camera model. I'd recommend taking a look at this notebook: https://github.com/facebookresearch/habitat-api/blob/master/notebooks/relative_camera_views_transform_and_warping_demo.ipynb

Hi, 

I checked the notebook, a question for intrinsic matrix, why the component point is located at (0, 0)?

K = np.array([
    [1 / np.tan(hfov / 2.), 0., 0., 0.],
    [0., 1 / np.tan(hfov / 2.), 0., 0.],
    [0., 0.,  1, 0],
    [0., 0., 0, 1]])

Usually the (0,0) of image coordinate is on the top-left, so is this different at habitat-api?

mosra, 2019-06-20:
Because the backend is running on OpenGL, the camera matrix is following the usual GL conventions, where (0, 0) is center and Y goes up.

placeforyiming, 2019-06-20:
> Because the backend is running on OpenGL, the camera matrix is following the usual GL conventions, where (0, 0) is center and Y goes up.

Got it, thanks!

placeforyiming, 2019-06-21:
> Because the backend is running on OpenGL, the camera matrix is following the usual GL conventions, where (0, 0) is center and Y goes up.

Hi @morsa,

Is there any way to get intrinsic matrix from habitat-sim (without api)?  Since now I can make habitat-sim work on Replica dataset, it almost can fulfill my need.

erikwijmans, 2019-06-21:
It can be gotten in the exact same way.  The default FOV is here: 
https://github.com/facebookresearch/habitat-sim/blob/bb87a9f1626d6f7d25a2513bf8dcb85123b9e5e2/src/esp/sensor/Sensor.h#L35-L37

You can change that to whatever value you want in a similar way to the changing of the sensor position in the stereo agent example:
https://github.com/facebookresearch/habitat-sim/blob/bb87a9f1626d6f7d25a2513bf8dcb85123b9e5e2/examples/stereo_agent.py#L67-L74

placeforyiming, 2019-06-21:
> It can be gotten in the exact same way. The default FOV is here:
>
> https://github.com/facebookresearch/habitat-sim/blob/bb87a9f1626d6f7d25a2513bf8dcb85123b9e5e2/src/esp/sensor/Sensor.h#L35-L37
>
> You [...]

Thank you all so much!

Have a good weekend!

closed 2019-06-21 by placeforyiming

QingquanBao, 2022-04-22:
Hello, since Habitat-sim experienced a big change in API, I wonder whether the [tutorial ](https://aihabitat.org/docs/habitat-api/view-transform-warp.html) still works. 

I have searched two files, [one python](https://github.com/facebookresearch/habitat-sim/blob/1ebba50749e44b7c64d7b464930dd9a10daf1391/examples/settings.py#L15) and [one C++](https://github.com/facebookresearch/habitat-sim/blob/61b3e721ba08f3dbfdc1dd3d9926b446e6eb3734/src/esp/sensor/CameraSensor.h#L17), where they say `hfov = 90`. So I wanna ensure that the default `hfov` equals to `90`?  I can still refer to the above tutorial?

By the way, is there any way to directly get the focal length of `CameraSensorSpec`, since I have not found the pixel density (which can then infer the focal length with the intrinsic params) ?

Finally but not important, I am using the class `ImageExtractor` and I feel struggled to find the intrinsics. So thanks u a lot if u can help me!

leobxpan, 2022-04-25:
Same question about intrinsics here. Would be great if we could confirm the default intrinsics parameters.

aclegg3, 2022-04-25:
Yes, `hfov = 90` is still the default, so above solutions still apply.

aclegg3, 2022-04-25:
I haven't run the code recently, but the math is still accurate on the tutorial linked above. Find more info on the doc page: https://aihabitat.org/docs/habitat-lab/view-transform-warp.html

leobxpan, 2022-04-25:
Thanks! this might be a dumb question - why's the focal length not in pixel units?

leobxpan, 2022-04-25:
Also, the parameters linked here https://github.com/facebookresearch/habitat-sim/issues/949 seem to have a different focal length / principle point.

erikwijmans, 2022-04-25:
The focal length is not specified in pixels so that's agnostic to the resolution of the camera. This is both nice for users (you can change resolution as you see fit) and nice for the renderer as OpenGL works in screen coordinate space, not pixel coordinate space.

The parameters in that file are different because it's using a different convention for specifying the intrinsics. (I forget which convention it's in because there are so many conventions for this...)

GuoPingPan, 2023-03-06:
> The focal length is not specified in pixels so that's agnostic to the resolution of the camera. This is both nice for users (you can change resolution as you see fit) and nice for the renderer as [...]

@erikwijmans 
Hi, if it is support the non-square camera to obtain the camear [Intrinsic parameters, K](https://aihabitat.org/docs/habitat-lab/view-transform-warp.html#intrinsic-parameters-k)?

chenhaomingbob, 2024-06-27:
@GuoPingPan 
Hi, I also need the intrinsic of a non-square camera. Do you find the way?