facebookresearch/habitat-sim #81: Install erro with setup.py

placeforyiming, 2019-06-19:
`[ 71%] Linking CXX executable stl_logging_unittest
[ 71%] Built target stl_logging_unittest
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
esp/core/CMakeFiles/core.dir/build.make:62: recipe for target 'esp/core/CMakeFiles/core.dir/esp.cpp.o' failed
make[2]: *** [esp/core/CMakeFiles/core.dir/esp.cpp.o] Error 4
CMakeFiles/Makefile2:5462: recipe for target 'esp/core/CMakeFiles/core.dir/all' failed
make[1]: *** [esp/core/CMakeFiles/core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
deps/assimp/code/CMakeFiles/assimp.dir/build.make:1700: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/Importer/IFC/IFCReaderGen2_2x3.cpp.o' failed
make[2]: *** [deps/assimp/code/CMakeFiles/assimp.dir/Importer/IFC/IFCReaderGen2_2x3.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
deps/corrade/src/Corrade/PluginManager/CMakeFiles/CorradePluginManager.dir/build.make:62: recipe for target 'deps/corrade/src/Corrade/PluginManager/CMakeFiles/CorradePluginManager.dir/AbstractManager.cpp.o' failed
make[2]: *** [deps/corrade/src/Corrade/PluginManager/CMakeFiles/CorradePluginManager.dir/AbstractManager.cpp.o] Error 4
CMakeFiles/Makefile2:3004: recipe for target 'deps/corrade/src/Corrade/PluginManager/CMakeFiles/CorradePluginManager.dir/all' failed
make[1]: *** [deps/corrade/src/Corrade/PluginManager/CMakeFiles/CorradePluginManager.dir/all] Error 2
[ 71%] Linking CXX executable example
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
deps/assimp/code/CMakeFiles/assimp.dir/build.make:1466: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/OgreXmlSerializer.cpp.o' failed
make[2]: *** [deps/assimp/code/CMakeFiles/assimp.dir/OgreXmlSerializer.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
deps/assimp/code/CMakeFiles/assimp.dir/build.make:1453: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/OgreBinarySerializer.cpp.o' failed
make[2]: *** [deps/assimp/code/CMakeFiles/assimp.dir/OgreBinarySerializer.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
deps/assimp/code/CMakeFiles/assimp.dir/build.make:1752: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/Importer/IFC/IFCProfile.cpp.o' failed
make[2]: *** [deps/assimp/code/CMakeFiles/assimp.dir/Importer/IFC/IFCProfile.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
deps/assimp/code/CMakeFiles/assimp.dir/build.make:972: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/ASELoader.cpp.o' failed
make[2]: *** [deps/assimp/code/CMakeFiles/assimp.dir/ASELoader.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
deps/assimp/code/CMakeFiles/assimp.dir/build.make:1193: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/LWOBLoader.cpp.o' failed
make[2]: *** [deps/assimp/code/CMakeFiles/assimp.dir/LWOBLoader.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
deps/assimp/code/CMakeFiles/assimp.dir/build.make:738: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/DeboneProcess.cpp.o' failed
make[2]: *** [deps/assimp/code/CMakeFiles/assimp.dir/DeboneProcess.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
deps/assimp/code/CMakeFiles/assimp.dir/build.make:114: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/ImporterRegistry.cpp.o' failed
make[2]: *** [deps/assimp/code/CMakeFiles/assimp.dir/ImporterRegistry.cpp.o] Error 4
c++: internal compiler error: Killed (program cc1plus)
[... 72 lines cut ...]
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/yzhao/.local/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/yzhao/.local/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/yzhao/.local/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/usr/lib/python3.6/distutils/command/install_lib.py", line 109, in build
    self.run_command('build_ext')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 166, in run
    self.build_extension(ext)
  File "setup.py", line 220, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2`

doyled-it, 2019-06-19:
Never mind. I saw [this closed issue](https://github.com/facebookresearch/habitat-sim/issues/25), and it worked for me.

placeforyiming, 2019-06-19:
Now I'm pretty sure this should be the bug in recent version. Since this error happens even in docker, and after I add 

> git checkout 62bc01be8cbc148eb099e96fa0b123e16b21f95f

It works well.

placeforyiming, 2019-06-19:
It even try to use my swap memory... Guess memory leaking in some place

mosra, 2019-06-19:
@placeforyiming how many CPU cores do you have? The compiler getting killed due to running out of memory is actually a quite possible scenario -- pybind11 python bindings code sometimes causes the compiler to consume *extreme* amounts of RAM, I encountered up to 2 GB per instance with GCC 8 and IIRC some older versions of GCC were consuming even more. 

In #38 we added an option to specify max number of parallel instances. Can you run `setup.py` with e.g. `--parallel 2`, if it helps anything?

placeforyiming, 2019-06-20:
I have 12 cores in my computer. The ubuntu system in my computer uses 1G RAM as swap memory which may cause this problem, guess hard disk is a more popular choice? I didn't install the system by myself, not sure how they did that part.   

I tried  `sudo python3 setup.py install  --parallel 2` and  `sudo python3 setup.py --parallel 2 install `
but neither worked. Guess this is not the right way to pass argument.

 I also see  `python setup.py build_ext --build-temp=build --parallel 3` from #38 , but didn't try this one, since I'm not sure if this is the same as `python setup.py install `.

Inspired by #24 , I changed 'j' to 'j6' in line 200, a little bit backward, but finally solved my problem.

closed 2019-06-20 by placeforyiming

mosra, 2019-06-20:
Yeah, it's a bit hard to get right: `--parallel` is an option of the `build_ext` command, so to do it all in one step you'd do this, I think:

```sh
python setup.py build_ext --parallel 3 install
```

Alternatively, <kbd>Ctrl</kbd> <kbd>C</kbd> during the build, `cd` into the `build/` directory and run either `make -jN` or `ninja -jN` there, then once that's done, resume `python setup.py install` -- it'll pick up the already built libraries. That's also generally the most flexible way to fiddle with CMake parameters and such.