facebookresearch/habitat-sim #82: Non-interactive testing by running the example script failed

Kyridiculous2, 2019-06-20:
## Steps to reproduce
I'm running on Ubuntu16.04.5 with python3.6.7
  python examples/example.py --scene /path/to/scene
## Observed Results
  Traceback (most recent call last):
  File "examples/example.py", line 12, in <module>
    import demo_runner as dr
  File "/data_1/codes/habitat-sim/examples/demo_runner.py", line 16, in <module>
    import habitat_sim
  File "/data_1/codes/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .nav import *
  File "/data_1/codes/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/data_1/codes/habitat-sim/habitat_sim/nav/greedy_geodesic_follower.py", line 6, in <module>
    import habitat_sim.agent
  File "/data_1/codes/habitat-sim/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/data_1/codes/habitat-sim/habitat_sim/agent/agent.py", line 12, in <module>
    import habitat_sim.bindings as hsim
  File "/data_1/codes/habitat-sim/habitat_sim/bindings/__init__.py", line 25, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ModuleNotFoundError: No module named 'habitat_sim._ext'

## Other observation
The Interactive testing works well

erikwijmans, 2019-06-20:
It seems like `/data_1/codes/habitat-sim` is in your `PYTHONPATH`  It shouldn’t be when habitat sim is installed.

Kyridiculous2, 2019-06-21:
Thanks a lot ! Solved this.

closed 2019-06-21 by Kyridiculous2

hzyjerry, 2019-06-25:
I encountered similar error on Mac:
```
Traceback (most recent call last):
  File "examples/example.py", line 12, in <module>
    import demo_runner as dr
  File "/Users/jerryhe/Desktop/Projects/habitat/habitat-sim/examples/demo_runner.py", line 16, in <module>
    import habitat_sim
  File "/Users/jerryhe/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-macosx-10.7-x86_64.egg/habitat_sim/__init__.py", line 12, in <module>
    from .nav import *
  File "/Users/jerryhe/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-macosx-10.7-x86_64.egg/habitat_sim/nav/__init__.py", line 1, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/Users/jerryhe/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-macosx-10.7-x86_64.egg/habitat_sim/nav/greedy_geodesic_follower.py", line 6, in <module>
    import habitat_sim.agent
  File "/Users/jerryhe/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-macosx-10.7-x86_64.egg/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/Users/jerryhe/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-macosx-10.7-x86_64.egg/habitat_sim/agent/agent.py", line 12, in <module>
    import habitat_sim.bindings as hsim
  File "/Users/jerryhe/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.0.1-py3.6-macosx-10.7-x86_64.egg/habitat_sim/bindings/__init__.py", line 25, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ModuleNotFoundError: No module named 'habitat_sim._ext'
```

What would be the recommended way to fix `PYTHONPATH`? Thanks for the help!

hzyjerry, 2019-06-25:
Ok I resolved on my end by `
export PYTHONPATH=$PYTHONPATH:/path/to/habitat-sim/`, hope this helps others who have the same issue.

balloch, 2019-08-16:
same problem, fix doesn't work for me. my Traceback:
```
Traceback (most recent call last):
  File "examples/example.py", line 12, in <module>
    import demo_runner as dr
  File "/home/balloch/software/habitat/habitat-sim/examples/demo_runner.py", line 16, in <module>
    import habitat_sim
  File "/home/balloch/software/habitat/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .nav import *
  File "/home/balloch/software/habitat/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/home/balloch/software/habitat/habitat-sim/habitat_sim/nav/greedy_geodesic_follower.py", line 6, in <module>
    import habitat_sim.agent
  File "/home/balloch/software/habitat/habitat-sim/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/home/balloch/software/habitat/habitat-sim/habitat_sim/agent/agent.py", line 14, in <module>
    import habitat_sim.bindings as hsim
  File "/home/balloch/software/habitat/habitat-sim/habitat_sim/bindings/__init__.py", line 24, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ModuleNotFoundError: No module named 'habitat_sim._ext'

```

Any ideas?

mathfac, 2019-11-12:
@balloch did you solve the issue? Your `PYTHONPATH` shouldn't include Habitat Sim dir.