facebookresearch/habitat-sim #86: make: *** No targets specified and no makefile found.  Stop.

goopci, 2019-06-24:
I'm on Ubuntu 18.04 with cmake 3.14.0 and I get the following output when running `python setup.py install`:

```
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is unknown
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at CMakeLists.txt:7 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/mthreet/habitat-sim/build/CMakeFiles/CMakeOutput.log".
See also "/home/mthreet/habitat-sim/build/CMakeFiles/CMakeError.log".
Traceback (most recent call last):
  File "setup.py", line 325, in <module>
    zip_safe=False,
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 166, in run
    self.build_extension(ext)
  File "setup.py", line 216, in build_extension
    env=env,
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-H/home/mthreet/habitat-sim/src', '-Bbuild', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/mthreet/habitat-sim/build/lib.linux-x86_64-3.7/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/home/mthreet/anaconda3/envs/hai/bin/python', '-DCMAKE_EXP
ORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=ON', '-DBUILD_TESTS=OFF']' returned non-zero exit status 1.
```

I then ran `export CXX=/usr/bin/cmake; python setup.py install` and got the following output:

```
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
make: *** No targets specified and no makefile found.  Stop.
Traceback (most recent call last):
  File "setup.py", line 325, in <module>
    zip_safe=False,
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 166, in run
    self.build_extension(ext)
  File "setup.py", line 220, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/home/mthreet/anaconda3/envs/hai/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.
```

Any idea what's going on?

erikwijmans, 2019-06-24:
The `CMAKE_CXX_COMPILER` is not `cmake`.  It is something like `g++` or `c++` or `clang++`.

You will also likely need to either clear the build folder `rm -rf build` or add `--force-cmake` to the end of your build command.

goopci, 2019-06-24:
Yep, after running `export CXX=/usr/gin/g++`  (and installing some missing dependencies) it was able to build.

closed 2019-06-24 by goopci