facebookresearch/habitat-sim #88: Can't find `download_mp.py`

nickgraz, 2019-06-26:
## Steps to reproduce

1. Clone repo (or open in Github)
2. Search for `download_mp.py` in repo

In the Readme, an urgent [update](https://github.com/facebookresearch/habitat-sim/blob/master/README.md#updates) is mentioned regarding the Matterport3D dataset, and to use the `download_mp.py` script to download it. It is also mentioned in the [Datasets](https://github.com/facebookresearch/habitat-sim/blob/master/README.md#datasets) section, and a command is given to run it. Where is this file? I am can't find it anywhere in the repo or any pull requests. Thanks!

[image]

erikwijmans, 2019-06-26:
`download_mp.py` is distributed by matterport. Please follow the instructions here: https://github.com/facebookresearch/habitat-sim#datasets

nickgraz, 2019-06-27:
I see, thanks!

closed 2019-06-27 by nickgraz