facebookresearch/habitat-sim #93: build --headless error

anoopsonar30, 2019-07-02:
## Steps to reproduce

  1. Run python setup.py install --headless

I also tried 
* python3 setup.py install --headless, 
* python3 setup.py install --headless --user,
* sudo python3 setup.py install --headless 
* sudo python setup.py install --headless 
* ./build.sh --headless

and they all result in the same error posted below.

## Observed Results

> Scanning dependencies of target datatool
> Scanning dependencies of target habitat_sim_bindings
> [ 97%] Building CXX object utils/datatool/CMakeFiles/datatool.dir/datatool.cpp.o
> [ 97%] Building CXX object utils/datatool/CMakeFiles/datatool.dir/__/__/deps/magnum/src/MagnumPlugins/AnyImageImporter/importStaticPlugin.cpp.o
> [ 98%] Building CXX object utils/datatool/CMakeFiles/datatool.dir/__/__/deps/magnum/src/MagnumPlugins/AnySceneImporter/importStaticPlugin.cpp.o
> make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libEGL.so', needed by 'utils/datatool/datatool'.  Stop.
> make[2]: *** Waiting for unfinished jobs....
> [ 98%] Building CXX object utils/datatool/CMakeFiles/datatool.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/StbImageImporter/importStaticPlugin.cpp.o
> [ 98%] Building CXX object utils/datatool/CMakeFiles/datatool.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/StbImageConverter/importStaticPlugin.cpp.o
> [ 98%] Building CXX object utils/datatool/CMakeFiles/datatool.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/AssimpImporter/importStaticPlugin.cpp.o
> [ 98%] Building CXX object utils/datatool/CMakeFiles/datatool.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/TinyGltfImporter/importStaticPlugin.cpp.o
> make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libEGL.so', needed by 'lib.linux-x86_64-3.6/habitat_sim/_ext/habitat_sim_bindings.cpython-36m-x86_64-linux-gnu.so'.  Stop.
> make[2]: *** Waiting for unfinished jobs....
> [ 98%] Building CXX object esp/bindings/CMakeFiles/habitat_sim_bindings.dir/__/__/deps/magnum/src/MagnumPlugins/AnySceneImporter/importStaticPlugin.cpp.o
> [ 98%] Building CXX object esp/bindings/CMakeFiles/habitat_sim_bindings.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/AssimpImporter/importStaticPlugin.cpp.o
> [ 98%] Building CXX object esp/bindings/CMakeFiles/habitat_sim_bindings.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/StbImageImporter/importStaticPlugin.cpp.o
> [ 99%] Building CXX object esp/bindings/CMakeFiles/habitat_sim_bindings.dir/__/__/deps/magnum/src/MagnumPlugins/AnyImageImporter/importStaticPlugin.cpp.o
> [ 99%] Building CXX object esp/bindings/CMakeFiles/habitat_sim_bindings.dir/bindings.cpp.o
> [ 99%] Building CXX object esp/bindings/CMakeFiles/habitat_sim_bindings.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/StbImageConverter/importStaticPlugin.cpp.o
> [100%] Building CXX object esp/bindings/CMakeFiles/habitat_sim_bindings.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/TinyGltfImporter/importStaticPlugin.cpp.o
> [100%] Building CXX object esp/bindings/CMakeFiles/habitat_sim_bindings.dir/geoBindings.cpp.o
> [100%] Building CXX object esp/bindings/CMakeFiles/habitat_sim_bindings.dir/ShortestPathBindings.cpp.o
> CMakeFiles/Makefile2:4795: recipe for target 'utils/datatool/CMakeFiles/datatool.dir/all' failed
> make[1]: *** [utils/datatool/CMakeFiles/datatool.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> CMakeFiles/Makefile2:4877: recipe for target 'esp/bindings/CMakeFiles/habitat_sim_bindings.dir/all' failed
> make[1]: *** [esp/bindings/CMakeFiles/habitat_sim_bindings.dir/all] Error 2
> Makefile:151: recipe for target 'all' failed
> make: *** [all] Error 2
> Traceback (most recent call last):
>   File "setup.py", line 325, in <module>
>     zip_safe=False,
>   File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/setuptools/__init__.py", line 129, in setup
>     return distutils.core.setup(**attrs)
>   File "/home/ubuntu/anaconda3/lib/python3.6/distutils/core.py", line 148, in setup
>     dist.run_commands()
>   File "/home/ubuntu/anaconda3/lib/python3.6/distutils/dist.py", line 955, in run_commands
>     self.run_command(cmd)
>   File "/home/ubuntu/anaconda3/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
>     self.do_egg_install()
>   File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
>     self.run_command('bdist_egg')
>   File "/home/ubuntu/anaconda3/lib/python3.6/distutils/cmd.py", line 313, in run_command
>     self.distribution.run_command(command)
>   File "/home/ubuntu/anaconda3/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
>     cmd = self.call_command('install_lib', warn_dir=0)
>   File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
>     self.run_command(cmdname)
>   File "/home/ubuntu/anaconda3/lib/python3.6/distutils/cmd.py", line 313, in run_command
>     self.distribution.run_command(command)
>   File "/home/ubuntu/anaconda3/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "/home/ubuntu/anaconda3/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
>     self.build()
>   File "/home/ubuntu/anaconda3/lib/python3.6/distutils/command/install_lib.py", line 107, in build
>     self.run_command('build_ext')
>   File "/home/ubuntu/anaconda3/lib/python3.6/distutils/cmd.py", line 313, in run_command
>     self.distribution.run_command(command)
>   File "/home/ubuntu/anaconda3/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "setup.py", line 166, in run
>     self.build_extension(ext)
>   File "setup.py", line 220, in build_extension
>     shlex.split("cmake --build {}".format(self.build_temp)) + build_args
>   File "/home/ubuntu/anaconda3/lib/python3.6/subprocess.py", line 291, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.

## Expected Results

  * Build successfully

## Relevant Details

gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.11)
I'm running this on an Amazon EC2 DLAMI instance.

erikwijmans, 2019-07-02:
> `make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libEGL.so', `

Looks like your AWS instance is miss libEGL.  You can install it via `sudo apt install libegl1-mesa-dev`

anoopsonar30, 2019-07-03:
It's already installed. Also there is no nvidia-opengl installation either.

erikwijmans, 2019-07-03:
Hmm, I am not sure how AWS sets up the DLAMI systems, so I am not sure what is missing from a normal nvidia-driver install.  This is the part of the  dockerbase we use for our docker containers that installs the OpenGL related things: https://gitlab.com/nvidia/opengl/blob/ubuntu18.04/glvnd/devel/Dockerfile

ankanbansal, 2019-07-08:
@anoopsonar30 Did you figure out a way to solve this error? I think I am getting basically the same error.

> ninja: error: '/usr/lib/x86_64-linux-gnu/libEGL.so', needed by 'utils/datatool/datatool', missing and no known rule to make it

mathfac, 2019-07-08:
Hi,
You can check our CI setup that installs Habitat on clean Ubuntu 16:
https://github.com/facebookresearch/habitat-api/blob/master/.circleci/config.yml#L45
for missing dependencies.

ankanbansal, 2019-07-08:
@mathfac Thank you for pointing to this. I still get the same error even after installing all the dependencies. I am using an Amazon EC2 DLAMI instance like @anoopsonar30

anoopsonar30, 2019-07-08:
@ankanbansal So I ended up using a plain Ubuntu 16.04 AMI installation on AWS instead of the DLAMI instance. Habitat doesn't seem to use CUDA. After that all you gotta do is install the latest NVIDIA graphics driver (I installed 430) and the dependencies in the install scripts as well as #1 and you're done.

ankanbansal, 2019-07-08:
Thank you, @anoopsonar30

closed 2019-11-26 by mathfac