facebookresearch/habitat-sim #94: Headless build GLXBadFBConfig error

anoopsonar30, 2019-07-03:
## Steps to reproduce

  1. Use habitat-docker on remote AWS server and set up X11 forwarding.
  2. (habitat) root@ip-172-31-77-104:/habitat-sim# python examples/example.py --scene ../habitat-api/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
 
## Observed Results

> ==== Initialized Sensor Spec: =====
> Sensor uuid:  color_sensor
> Sensor type:  SensorType.COLOR
> Sensor position:  [0.  1.5 0. ]
> ===================================
> debug1: client_input_channel_open: ctype x11 rchan 3 win 65536 max 16384
> debug1: client_request_x11: request from 172.31.77.104 57626
> debug1: channel 1: new [x11]
> debug1: confirm x11
> X Error of failed request:  GLXBadFBConfig
>   Major opcode of failed request:  149 (GLX)
>   Minor opcode of failed request:  34 ()
>   Serial number of failed request:  21
>   Current serial number in output stream:  22
> debug1: channel 1: FORCE input drain

## Expected Results

Something like - 
640 x 480, total time: 3.208 sec. FPS: 311.7

## Relevant Code

I have received the expected output in the past. However I'm suddenly getting this error. I don't think this is an X forwarding error since other applications like xclock and xeyes seem to be working just fine.

anoopsonar30, 2019-07-05:
@erikwijmans

erikwijmans, 2019-07-06:
How X11 forwarding works is quite strange and it seems like the OpenGL specs after OpenGL 2 didn't bother implementing it, so it may never work as we rely in features from these version.  If `glxinfo` isn't working, there is definitely nothing we can do to fix that on our end.

anoopsonar30, 2019-07-07:
I see. That's unfortunate. I guess I'll try using VNC and see if that works.

aclegg3, 2022-08-31:
Seems like limitations were identified and the issue can be closed.

closed 2022-08-31 by aclegg3