facebookresearch/habitat-sim #95: Build issue: AttributeError: 'dict' object has no attribute '__NUMPY_SETUP__'

ToniRV, 2019-07-03:
## Steps to reproduce

  1. Clone repo
  2. Fresh virtualenv: `virtualenv -p python3.6 ./venv && source ./venv/bin/activate`
  3. Install numpy: `pip install numpy`

## Observed Results

  * What happened? Compilation issue in Ubuntu 18.04 and compiler:
```
/usr/lib/ccache/gcc
gcc --version
gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
```
Log output:
```
Installed /home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/Pillow-6.1.0-py3.6-linux-x86_64.egg
Searching for numpy-quaternion
Reading https://pypi.org/simple/numpy-quaternion/
Downloading https://files.pythonhosted.org/packages/2e/60/5f944c433e3726a340db9cd0ef24bd4ac34c4d3654ca8dc8f9cc0e21ce78/numpy-quaternion-2019.6.26.15.8.24.tar.gz#sha256=5b11f65f3ef6fa0227530d2c587cc0cf57f64a46d5a121e1cf99adc4c2807014
Best match: numpy-quaternion 2019.6.26.15.8.24
Processing numpy-quaternion-2019.6.26.15.8.24.tar.gz
Writing /tmp/easy_install-cftr53xo/numpy-quaternion-2019.6.26.15.8.24/setup.cfg
Running numpy-quaternion-2019.6.26.15.8.24/setup.py -q bdist_egg --dist-dir /tmp/easy_install-cftr53xo/numpy-quaternion-2019.6.26.15.8.24/egg-dist-tmp-5t0ojvyj
The variable 'package_version' was not present in the environment
Traceback (most recent call last):
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/sandbox.py", line 154, in save_modules
    yield saved
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/sandbox.py", line 195, in setup_context
    yield
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/sandbox.py", line 250, in run_setup
    _execfile(setup_script, ns)
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/sandbox.py", line 45, in _execfile
    exec(code, globals, locals)
  File "/tmp/easy_install-cftr53xo/numpy-quaternion-2019.6.26.15.8.24/setup.py", line 117, in <module>
    def __init__(self, name, sourcedir=""):
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 163, in run
    self.run_command("egg_info")
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/egg_info.py", line 296, in run
    self.find_sources()
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/egg_info.py", line 303, in find_sources
    mm.run()
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/egg_info.py", line 534, in run
    self.add_defaults()
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/egg_info.py", line 570, in add_defaults
    sdist.add_defaults(self)
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/py36compat.py", line 36, in add_defaults
    self._add_defaults_ext()
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/py36compat.py", line 119, in _add_defaults_ext
    build_ext = self.get_finalized_command('build_ext')
  File "/usr/lib/python3.6/distutils/cmd.py", line 299, in get_finalized_command
    cmd_obj.ensure_finalized()
  File "/usr/lib/python3.6/distutils/cmd.py", line 107, in ensure_finalized
    self.finalize_options()
  File "/tmp/easy_install-cftr53xo/numpy-quaternion-2019.6.26.15.8.24/setup.py", line 109, in finalize_options
    
AttributeError: 'dict' object has no attribute '__NUMPY_SETUP__'

[... 79 lines cut ...]
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/egg_info.py", line 296, in run
    self.find_sources()
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/egg_info.py", line 303, in find_sources
    mm.run()
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/egg_info.py", line 534, in run
    self.add_defaults()
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/egg_info.py", line 570, in add_defaults
    sdist.add_defaults(self)
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/py36compat.py", line 36, in add_defaults
    self._add_defaults_ext()
  File "/home/tonirv/Code/Simulators/habitat-sim/venv/lib/python3.6/site-packages/setuptools/command/py36compat.py", line 119, in _add_defaults_ext
    build_ext = self.get_finalized_command('build_ext')
  File "/usr/lib/python3.6/distutils/cmd.py", line 299, in get_finalized_command
    cmd_obj.ensure_finalized()
  File "/usr/lib/python3.6/distutils/cmd.py", line 107, in ensure_finalized
    self.finalize_options()
  File "/tmp/easy_install-cftr53xo/numpy-quaternion-2019.6.26.15.8.24/setup.py", line 109, in finalize_options
    
AttributeError: 'dict' object has no attribute '__NUMPY_SETUP__'

```

## Expected Results

  * What did you expect to happen? Successful install...

erikwijmans, 2019-07-03:
This seems to be a build issue with one of the dependencies, https://github.com/moble/quaternion.  Can you open an issue there?

You can also see if installing it directly via `pip install numpy numpy-quaternion` works.

ToniRV, 2019-07-03:
Ok, thanks!

closed 2019-07-03 by ToniRV