facebookresearch/habitat-sim #96: Placing an agent into an Imported mesh in Habitat

medhagitnod, 2019-07-03:
I have imported a mesh into Habitat-SIM and am able to use build/viewer to confirm that it is the correct version of glb file. The mesh is a cubical closed room, and I would like to place an agent inside the mesh itself to explore the room. What would be the best way to do this? Is there a way to get the coordinates of the mesh itself and be able to manually place an agent inside it?

erikwijmans, 2019-07-04:
Probably the simplest way is to create a navmesh via the `datatool`, `build/utils/datatool/datatool`: https://github.com/facebookresearch/habitat-sim/blob/master/src/utils/datatool/datatool.cpp
i.e.

```
build/utils/datatool/datatool create_navmesh <mesh_name>.glb <mesh_name>.navmesh
```
You'll have to make sure the mesh has its gravity direction along -Y.

Then you can sample random navigation locations via `sim.pathfinder.get_random_navigable_point()`.

ronjamoller, 2020-02-11:
@erikwijmans sorry to barge in 6 months after this was asked - but I am facing the same problem. How does one compile the datatool properly now ? I cant find documentation on how to compile it  and only the viewer is compiled by default - and since a lot has changed since july i figured i would ask first.
thank you !

erikwijmans, 2020-02-11:
`python setup.py --help` or `./build.sh --help` shows the various compile args.  You can compile it by passing `--build-datatool`

ronjamoller, 2020-02-26:
Thank you for the help - i have successfully created the navmesh for a simple box mesh from blender that occupies roughly the same volume, footprint and orintation as the sokloster test scene, now i have the following issues: 
1. the generated navmesh only allows me to stand in place and turn, i cannot move. If i simply copy the sokloster navmesh and copy and rename it so it matches my demo room i can navigate in the demo room but not with the navmesh that is generated with the datatool (the output of the datatool does not tell me that anything is wrong, unlike when i had an actual faulty mesh). 
2. Furthermore, are there any guidelines for lighting / emission settings that are recommended because while my inserted objects are lit properly and when i look into the box in blender it is properly textured it does show up as solid black in habitat.
[image]

Thank you for the help.

ronjamoller, 2020-02-27:
I have fixed the texture issue, it was a blender rendering problem - the navmesh problem persists unfortunately, if anyone has ideas I would be very thankful. I would like to avoid having to use the sokloster navmesh as there are occlusions like the center table and the chairs.
thank you

erikwijmans, 2020-02-27:
What direction is gravity in your box mesh?  As .glb meshes from gibson/matterport3d/sokloster are all -Z gravity, we have logic that assumes that these scene-level meshes are -Z gravity.

ronjamoller, 2020-03-01:
thank you so much for the hint - it was the gravity issue, i am relatively new to blender so i didnt know there was an explicit gravity setting and assumed it was just about the orientation of the mesh (which i compared to the matterport and test scenes and since they faced the same direction i thought it was fine). For anyone going the blender route as well - the gravity has to be along the positive y axis in blender. 
Thank you !

closed 2021-03-01 by dhruvbatra