facebookresearch/habitat-sim #97: Variable move_forward distances

anoopsonar30, 2019-07-07:
What would be an easy way to have the agent move forward by variable distances during each timestep? 

Normally we have to declare the actions in the action_space like this - 

```
    agent_cfg.action_space = {
        "move_forward": habitat_sim.agent.ActionSpec(
            "move_forward", habitat_sim.agent.ActuationSpec(amount=0.25)
        ),
        "turn_left": habitat_sim.agent.ActionSpec(
            "turn_left", habitat_sim.agent.ActuationSpec(amount=30.0)
        ),
        "turn_right": habitat_sim.agent.ActionSpec(
            "turn_right", habitat_sim.agent.ActuationSpec(amount=30.0)
        ),
    }
```

where the ActuationSpec amounts are fixed. What would be an efficient way to make it such that the agent moves/turns different distances after each timestep?

Also, how can one detect collisions between the Agent and the environment?

@erikwijmans Thanks!

erikwijmans, 2019-07-08:
- Variable move distances: The `amount` the agent moves is just a value passed to the function that implements the action every time, so you can change that part of the agents config on-the-fly and it will be reflected. i.e.
```
agent.agent_config.action_space[<action_name>].actuation.amount = <new_amount>
```

- Detection collisions:  The simulator returns whether the action resulted in a collision or not in the observations:
https://github.com/facebookresearch/habitat-sim/blob/5fb97d2014d00078e1e772acbd2af9e637b7db1a/habitat_sim/simulator.py#L166

The agents act method also returns a boolean indicating collision or not:
https://github.com/facebookresearch/habitat-sim/blob/5fb97d2014d00078e1e772acbd2af9e637b7db1a/habitat_sim/agent/agent.py#L153-L162

anoopsonar30, 2019-07-11:
@erikwijmans 
I'm trying to add "move_upwards" as a body action so that it moves the entire agent and not just the sensors. However adding - 

`register_move_fn(MoveUp, name="move_upwards", body_action=True)`

to default_controls.py did not make it work. The program runs without errors but does nothing and does not change the position of the agent.

anoopsonar30, 2019-07-11:
It seemed like a pain so I just moved it up like this - 

`initState.position[1] = initState.position[1] + 1.5`

erikwijmans, 2019-07-11:
Currently, something like `move_up` cannot be a body action as body actions are forced to always result in a position that is on the navigation mesh (i.e. on some navigable surface like the floor).  This can be changed by introducing an additional parameter such that `apply_filter` is not always `True` like it is currently for body actions:
https://github.com/facebookresearch/habitat-sim/blob/eb191d19165a32101a3745bde03518ddfe86155e/habitat_sim/agent/agent.py#L158

anoopsonar30, 2019-07-12:
Ah I see. I'm trying to simulate a quadrotor so I just did it like I said in my previous comment and then set the agent's state to the new edited state. May not be the most elegant way to do it but it works I guess.

anoopsonar30, 2019-07-12:
But how does this influence the collision detection code? Will it still work correctly? I mean, does obs['collided'] still return the right value?

erikwijmans, 2019-07-12:
The collisions unfortunately won't really work.  The current collision code doesn't support a flying agent, i.e. the quadrotor won't be able to fly over obstacles.

anoopsonar30, 2019-07-22:
Do you expect that this feature will be added soon? @erikwijmans

erikwijmans, 2019-07-22:
Something like a quadrotor will have to be simulated via full physics simulation.  Follow along with that progress here: https://github.com/facebookresearch/habitat-sim/pull/91

Danmou, 2020-01-30:
Now that physics is here, is supporting full 3D movement on the roadmap?

erikwijmans, 2020-01-30:
@aclegg3 I believe you have worked on something like this previously?

aclegg3, 2020-01-30:
Hey, @Danmou. I'd be happy to hear your use case. 

With the existing functionality you can command positional and rotational kinematic motion as well as forces and torques for rigid bodies. A supported method for attaching an Agent to a simulated object is also in the works, but you can already manually sync the agent's transformation with that of a RigidObject (as discussed in another thread #345). 

It is currently on the roadmap to enable velocity control for the current rigid body integration. It is also on the roadmap to incorporate URDF and multi-body (articulated rigid body) physics with joint|motor velocity|torque control.

Danmou, 2020-01-30:
Hi @aclegg3, thanks for the update.

Actually, the simple quadcopter use-case discussed in the thread you linked was exactly what I had in mind. I think I'll wait until there's a more built-in solution though, as I have more pressing issues at the moment. Velocity control and dynamics are not too much of a priority for me, but it would be nice to have proper support for continuous action spaces (currently I'm just changing the amount on the fly).

It's gonna be exciting to see support for multi-body systems and general physics-based control. One thing that would be nice if you could keep in mind when implementing this is support for domain randomization (e.g. having a different randomized mass for each body in each episode) and external disturbances.

aclegg3, 2020-01-31:
Great point about domain randomization. Our current design for rigid body simulation is to load assets into a `ResourceManager` and instance them in the world by referencing an object template. By modifying the template parameters (such as mass, scale, friction, damping, etc...) before adding objects you can apply domain randomization. We'll keep this in mind while developing our multi-body support.

aclegg3, 2022-02-03:
Closing this as it is no longer relevant. 

Continuous actions are supported in Habitat-lab as are simulator actions that directly leverage the Habitat-sim API. This means you can modify the Agent state however you wish. VelocityControl and force/torque control are also supported through this approach.

closed 2022-02-03 by aclegg3