facebookresearch/habitat-sim #98: path separator issue on microsoft winsuck

AlonzoTG, 2019-07-08:
My linux machine is overcommitted right now so I'm trying to test this on a Micro$oft Winsuck 10 machine. 

I installed Anaconda as my python environment and am using the anaconda-ized command prompt to use this. 

As the console output below shows, the thing is either trying to use unix style path seperators or no separators at all, machine of course requires DOS style separators... =\

####
(base) C:\Users\atg\Documents\GitHub\habitat-sim>python setup.py install
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info\PKG-INFO
writing dependency_links to habitat_sim.egg-info\dependency_links.txt
writing requirements to habitat_sim.egg-info\requires.txt
writing top-level names to habitat_sim.egg-info\top_level.txt
reading manifest file 'habitat_sim.egg-info\SOURCES.txt'
writing manifest file 'habitat_sim.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
CMake Error: The source directory "C:UsersatgDocumentsGitHubhabitat-simsrc" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Traceback (most recent call last):
  File "setup.py", line 325, in <module>
    zip_safe=False,
  File "C:\Users\atg\Anaconda3\lib\site-packages\setuptools\__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "C:\Users\atg\Anaconda3\lib\distutils\core.py", line 148, in setup
    dist.run_commands()
  File "C:\Users\atg\Anaconda3\lib\distutils\dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "C:\Users\atg\Anaconda3\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "C:\Users\atg\Anaconda3\lib\site-packages\setuptools\command\install.py", line 67, in run
    self.do_egg_install()
  File "C:\Users\atg\Anaconda3\lib\site-packages\setuptools\command\install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "C:\Users\atg\Anaconda3\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\atg\Anaconda3\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "C:\Users\atg\Anaconda3\lib\site-packages\setuptools\command\bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "C:\Users\atg\Anaconda3\lib\site-packages\setuptools\command\bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "C:\Users\atg\Anaconda3\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\atg\Anaconda3\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "C:\Users\atg\Anaconda3\lib\site-packages\setuptools\command\install_lib.py", line 11, in run
    self.build()
  File "C:\Users\atg\Anaconda3\lib\distutils\command\install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "C:\Users\atg\Anaconda3\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\atg\Anaconda3\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 166, in run
    self.build_extension(ext)
  File "setup.py", line 216, in build_extension
    env=env,
  File "C:\Users\atg\Anaconda3\lib\subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-HC:UsersatgDocumentsGitHubhabitat-simsrc', '-BbuildRelease', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=C:\\Users\\atg\\Documents\\GitHub\\habitat-sim\\build\\lib.win-amd64-3.7\\habitat_sim\\_ext', '-DPYTHON_EXECUTABLE=C:\\Users\\atg\\Anaconda3\\python.exe', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=ON', '-DBUILD_TESTS=OFF']' returned non-zero exit status 1.

(base) C:\Users\atg\Documents\GitHub\habitat-sim>

dhruvbatra, 2019-07-08:
Windows isn't supported till #84 lands. Please follow progress there.

aclegg3, 2022-03-01:
No planned windows support at this time.

closed 2022-03-01 by aclegg3