facebookresearch/habitat-sim #100: --save_png saves all black images

iszihan, 2019-07-15:
Hello, I am trying to render out images from a Replica scene with the --save_png option. However, the output png files are all-black/gray (with rgb value (31,31,31)). I tried it again with test scene also, the output images are still gray. I am wondering what could be the possible reason for this and how could I fix it? Thank you!

erikwijmans, 2019-07-15:
These are working for me locally and on the CI (see https://github.com/facebookresearch/habitat-sim/blob/master/tests/test_sensors.py).  If this is a machine with a display attached, does the interactive viewer work?

PsycoTodd, 2019-07-15:
You may also load the wrong ply file. Try the one in the habitat directory. BTW, what is the command you use to run the image saving?

iszihan, 2019-07-15:
@erikwijmans Yes, the interactive viewer is working. I'm using the example.py

iszihan, 2019-07-15:
@PsycoTodd for replica dataset, i loaded the one in habitat folder named mesh_semantics.ply. The command I used is "python example.py --scene path/to/the/ply --save_png "

PsycoTodd, 2019-07-16:
@iszihan I use python3 examples/example.py --scene <path_replica>/<room_0>/habitat/mesh_semantic.ply --save_png --semantic_sensor, you can also add --max_frame 20 to avoid get too much image rendered.

iszihan, 2019-07-19:
it works now on linux. not sure what the issue is with mac but ill close it for now :) thank you

closed 2019-07-19 by iszihan