facebookresearch/habitat-sim #101: Rotation axis seems incorrect for turn_left and turn_right

aphalak, 2019-07-15:
The implementation for turning is implemented in default_controls.py as :
`register_move_fn(LookLeft, name="turn_left", body_action=True)
register_move_fn(LookRight, name="turn_right", body_action=True)`

You would expect look_left and turn_left to render the same image, however turn_left seems like rotates the agent and/or camera about an incorrect vertical axis.
To reproduce the problem, you can try to keep turning left repeatedly. Ideally you would expect the camera to render the initial image after a 360 turn, but in reality the agent spins out of control and out of the mesh

erikwijmans, 2019-07-15:
Can you provide a reproduction script?  I can't reproduce this locally with 
```
import tempfile

import cv2
import imageio
import numpy as np

import habitat_sim

np.random.seed(10)

FORWARD_KEY = "w"
LEFT_KEY = "a"
RIGHT_KEY = "d"
DONE_KEY = "q"

def transform_rgb_bgr(image):
    return image[:, :, [2, 1, 0]]

def example():
    agent_cfg = habitat_sim.AgentConfiguration()
    agent_cfg.sensor_specifications[0].resolution = [1024, 1024]
    agent_cfg.sensor_specifications[0].position = [0.0, 1.0, 0.0]
    agent_cfg.action_space["look_left"] = habitat_sim.ActionSpec(
        "look_left", habitat_sim.ActuationSpec(10.0)
    )
    agent_cfg.action_space["look_right"] = habitat_sim.ActionSpec(
        "look_right", habitat_sim.ActuationSpec(10.0)
    )

    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.scene.id = (
        "./data/scene_datasets/replica_dataset/apartment_0/habitat/mesh_semantic.ply"
    )
    sim = habitat_sim.Simulator(habitat_sim.Configuration(sim_cfg, [agent_cfg]))

    state = habitat_sim.AgentState()
    state.position = np.array([1.460192, -1.3747652, -1.5571239], dtype=np.float32)
    state.rotation = np.quaternion(-0.31170254945755, 0, -0.950179815292358, 0)
    sim.get_agent(0).state = state

    obs = sim.get_sensor_observations()
    cv2.imshow("RGB", transform_rgb_bgr(obs["rgba_camera"]))

    print("Agent stepping around inside environment.")

    while True:
        k = cv2.waitKey()
        if k == ord(FORWARD_KEY):
            act = "move_forward"
        elif k == ord(LEFT_KEY):
            act = "turn_left"
        elif k == ord(RIGHT_KEY):
            act = "turn_right"
        elif k == ord("l"):
            act = "look_left"
        elif k == ord("r"):
            act = "look_right"
        elif k == ord(DONE_KEY):
            break
        else:
            continue

        obs = sim.step(act)
        cv2.imshow("RGB", transform_rgb_bgr(obs["rgba_camera"]))

if __name__ == "__main__":
    example()
```

aphalak, 2019-07-15:
I see, your example seems to work as expected for me too. Maybe there's some bug in the scripts on my side. I can start off from your script and we can mark this as resolved. For the sake of completion, here is what my attempt looked like : 
I modified the demo_runner.py scripts to start off

**manual_runner.py**

`
import multiprocessing
import os
import random
import sys
import time
from enum import Enum

import numpy as np
import cv2
from PIL import Image

import habitat_sim
import habitat_sim.agent
import habitat_sim.bindings as hsim
import habitat_sim.utils as utils
from settings import default_sim_settings, make_cfg

_barrier = None

def mapKeyToAction(key):
    key = chr(key)
    if key == 'w':
        return "move_forward"
    if key == 's':
        return "move_backward"
    if key == 'a':
        return "turn_left"
    if key == 'l':
        return "look_left"
    if key == 'd':
        return "turn_right"
    if key == 'o':
        return "look_up"
    if key == 'p':
        return "look_down"

    return None

class DemoRunnerType(Enum):
    BENCHMARK = 1
    EXAMPLE = 2

class DemoRunner:
    def __init__(self, sim_settings, simulator_demo_type):
        if simulator_demo_type == DemoRunnerType.EXAMPLE:
            self.set_sim_settings(sim_settings)

    def set_sim_settings(self, sim_settings):
        self._sim_settings = sim_settings.copy()

    def save_color_observation(self, obs, total_frames):
        color_obs = obs["color_sensor"]
        color_img = Image.fromarray(color_obs, mode="RGBA")
        color_img.save("test.rgba.%05d.png" % total_frames)

    def save_semantic_observation(self, obs, total_frames):
        semantic_obs = obs["semantic_sensor"]
        semantic_img = Image.new("P", (semantic_obs.shape[1], semantic_obs.shape[0]))
        semantic_img.putpalette(utils.d3_40_colors_rgb.flatten())
        semantic_img.putdata((semantic_obs.flatten() % 40).astype(np.uint8))
        semantic_img.save("test.sem.%05d.png" % total_frames)

    def save_depth_observation(self, obs, total_frames):
        if self._sim_settings["depth_sensor"]:
[... 209 lines cut ...]
`
And in settings.py, I set these controls :
`agent_cfg.action_space = {
        "move_forward": habitat_sim.agent.ActionSpec(
            "move_forward", habitat_sim.agent.ActuationSpec(amount=0.25)
        ),
        "move_backward": habitat_sim.agent.ActionSpec(
            "move_backward", habitat_sim.agent.ActuationSpec(amount=0.25)
        ),
        "turn_left": habitat_sim.agent.ActionSpec(
            "turn_left", habitat_sim.agent.ActuationSpec(amount=5.0)
        ),
        "look_left": habitat_sim.agent.ActionSpec(
            "look_left", habitat_sim.agent.ActuationSpec(amount=5.0)
        ),
        "turn_right": habitat_sim.agent.ActionSpec(
            "turn_right", habitat_sim.agent.ActuationSpec(amount=5.0)
        ),
        "look_up": habitat_sim.agent.ActionSpec(
            "look_up", habitat_sim.agent.ActuationSpec(amount=10.0)
        ),
        "look_down": habitat_sim.agent.ActionSpec(
            "look_down", habitat_sim.agent.ActuationSpec(amount=10.0)
        ),
    }`

erikwijmans, 2019-07-15:
`look_{up, down}` and `look_{left, right}` can have a weird interaction with each-other as they do change each-others rotational coordinate frame, but the agent shouldn't be able to escape the mesh