facebookresearch/habitat-sim #102: How to label certain category of primitives in the semantic image?

PsycoTodd, 2019-07-15:
## Steps to reproduce

  1. Change the code that loads the primitive id. If the id == let's say floor's id (40, get from the semantic.json file), pass the id to the texture generation procedure, or else pass id = 0 or a hard coded value.
  2. compile the tool
  3. run with a replica dataset.

## Observed Results
The whole semantic result is a dark blue image, even the RGB result shows indeed floor inside the image. 
  * What happened?  This could be a description, log output, etc.

## Expected Results
I expect the all the wall will be one color while the other parts be in a solid color in the semantic png result.
  * What did you expect to happen?

## Relevant Code
https://github.com/facebookresearch/habitat-sim/blob/eb191d19165a32101a3745bde03518ddfe86155e/src/esp/assets/GenericInstanceMeshData.cpp#L187
```
// TODO(you): code here to reproduce the problem
```
    for (auto id : tmp) {
     if(id != 40) {
         id = 0;
      }
      objectIds_.emplace_back(id);
      if (vertexPerFace == 4)
        objectIds_.emplace_back(id);
    }

erikwijmans, 2019-07-15:
The ID's in the mesh are the _instance_ ID's, not the semantic class ID's.

PsycoTodd, 2019-07-16:
Does this means the generated image the mesh color represent instance not semantics?
So is that possible to visualize certain types in term of semantics in the room model?

Thank you.

erikwijmans, 2019-07-16:
For the replica dataset, there is mapping between the instance IDs and the semantic label IDs already in the annotations.  You can get it like
```
import json
import numpy as np

with open("./apartment_0/habitat/info_semantic.json", "r") as f:
    annotations = json.load(f)

instance_id_to_semantic_label_id = np.array(annotations["id_to_label"])
```

Then you can do
```
sem_seg = instance_id_to_semantic_label_id[obs['semantic']]
```

PsycoTodd, 2019-07-16:
Hi Erik
Your suggestion works. Thanks for pointing me out to the correct direction.

Todd

closed 2019-07-16 by PsycoTodd