facebookresearch/habitat-sim #115: Clang compile failure

mpiseno, 2019-07-19:
## Steps to reproduce

 1. git clone https://github.com/facebookresearch/habitat-sim.git
2. conda activate habitat (which include python3)
3. cd habitat-sim
4. pip install -r requirements.txt
5. python setup.py install --user

## Observed Results
Error message:

********************
make[2]: *** [esp/scene/CMakeFiles/scene.dir/Mp3dSemanticScene.cpp.o] Error 254
clang: error: unable to execute command: Illegal instruction: 4
clang: error: clang frontend command failed due to signal (use -v to see invocation)
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
Target: x86_64-apple-darwin18.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
clang: note: diagnostic msg: PLEASE submit a bug report to http://developer.apple.com/bugreporter/ and include the crash backtrace, preprocessed source, and associated run script.
clang: note: diagnostic msg: 
********************

macOS: 10.14.5
xCode: 10.2.1
clang: 1001.0.46.4
python: 3.6

erikwijmans, 2019-07-19:
IIRC, this happens when you run out of memory.

Try
`python setup.py build_ext --parallel 1 install`

ChanganVR, 2019-09-02:
I met the same problem and adding the parallel option didn't work for me. Below is error message I had: 

[84%] Building CXX object esp/physics/CMakeFiles/physics.dir/PhysicsManager.cpp.o
clang: error: unable to execute command: Illegal instruction: 4
clang: error: clang frontend command failed due to signal (use -v to see invocation)
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
clang: note: diagnostic msg: PLEASE submit a bug report to http://developer.apple.com/bugreporter/ and include the crash backtrace, preprocessed source, and associated run script.
clang: note: diagnostic msg:
********************

ChanganVR, 2019-09-04:
I rolled the codes out to the last stable release and it was fixed.

cesleem, 2020-01-28:
Hi, I'm getting this same error and I've tried the stable build and previous build (v1.3).

Tried both: `python setup.py install` and `python setup.py build_ext --parallel 1 install`

#### Error Message:
---
Scanning dependencies of target physics
[ 81%] Building CXX object esp/physics/CMakeFiles/physics.dir/PhysicsManager.cpp.o
clang: error: unable to execute command: Illegal instruction: 4
clang: error: clang frontend command failed due to signal (use -v to see invocation)
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
clang: note: diagnostic msg: PLEASE submit a bug report to http://developer.apple.com/bugreporter/ and include the crash backtrace, preprocessed source, and associated run script.
clang: note: diagnostic msg:

...

make[2]: *** [esp/physics/CMakeFiles/physics.dir/PhysicsManager.cpp.o] Error 254
make[1]: *** [esp/physics/CMakeFiles/physics.dir/all] Error 2
make: *** [all] Error 2

---

macOS: 10.14.6 (Mojave)
xCode: 11.2
clang: 1001.0.46.4
python: 3.6.10

Not sure how stable the conda pkg version of habitat-sim is but I'm happy to test that as an alternative.

cesleem, 2020-01-28:
Update that I was able to get a conda install working using `conda install -c aihabitat -c conda-forge habitat-sim` from [https://github.com/facebookresearch/habitat-sim/tree/conda-build-2/conda-build](https://github.com/facebookresearch/habitat-sim/tree/conda-build-2/conda-build).

dhruvbatra, 2020-01-29:
Glad you got conda install to work. It's highly experimental and under active development (you can follow along in #445). But eventually, it will make everyone's life easier.

cesleem, 2020-01-29:
Thank you for the pointer!  Agreed, thanks to the team! Very excited to see this option, beats 2-3 hours of debugging!

closed 2021-03-01 by dhruvbatra