facebookresearch/habitat-sim #121: Proposal: replace glog with Corrade::Utility::Debug

mosra, 2019-07-27:
I'm seeing a lot of code like

```cpp
LOG(INFO) << "User provided object inertia " 
          << bInertia.x() << " " << bInertia.y() << " " << bInertia.z();
```

or a `switch` that first converts an enum to a string and then prints it with `LOG`. It's making me sad, because it wastes programmer time -- logging / printf debugging should not involve any mental overhead of "how the heck do I print this variable". The [Utility::Debug](https://doc.magnum.graphics/corrade/classCorrade_1_1Utility_1_1Debug.html) class in Magnum can do the same as just:

```cpp
Debug{} << "User provided object inertia" << bInertia;
```

and it can also print *any* container, almost all Magnum enum values (as names, not numbers, of course), it has output redirection (and muting), debug/warning/error levels, colored output if you want to get fancy and it can [print images](https://blog.magnum.graphics/backstage/multidimensional-strided-array-views/#enter-strided-array-views) as well.

I never used glog, but looking at the project page there doesn't even seem to be any real docs for it, apart from a [raw HTML file](https://github.com/google/glog/blob/master/doc/glog.html). How well maintained that project is? For the WebGL build in #53 we'd need to use something else anyway, so I would propose switching away from it completely.

My question is -- is there any glog feature that `Utility::Debug` doesn't have and you need it? One thing I'm aware of is that it doesn't prefix the output with file/line by default, but there's a (not yet merged) proposal that adds it as an opt-in feature on supported compilers: https://github.com/mosra/corrade/pull/55

msavva, 2019-07-30:
Thanks @mosra for making this proposal!  It will certainly make things easier for us down the line if we can leverage logging from Magnum (and get free WebGL build support).  Overall, it does seem that switching from `glog` to `Utility::Debug` will bring benefits without giving up too much.

In addition to file/line prefixes, a few other features of `glog` that we currently use are: having a file sink for the log, verbosity level control and module-level log output filtering through "setting flags" (see http://rpg.ifi.uzh.ch/docs/glog.html).  Are analogues of these features easy through `Utility::Debug`?

mosra, 2019-07-30:
There's `Debug`, `Warning`, `Error` and `Fatal` (which exits after printing a message) for different verbosity levels. Each of them can be separately scope-redirected to any of C++ streams (so stdout, stderr, file output, ...) and also muted (see the docs linked above for details). Redirecting to standard output *and* a file would need to be done using a custom `ostream` implementation, at the moment there's no such functionality. 

If you'd need to configure the file redirection / muting / verbosity level through command-line arguments or environment, you would need to wire that in (for example using `Utility::Arguments`).
Compared to glog it's completely macro-less, which means some features that glog has can't be easily done without wrapping everything in macros as well. That means it also affects everything related to file/line info (such as filename-based filtering) -- mosra/corrade#55 relies on `std::source_location` to get this info and compiler support is pretty spotty at the moment.

In my experience I needed file/line info very rarely in practice and definitely not for every message, but I don't want to force you into anything :)

mosra, 2019-08-30:
#179 might be another reason to move away from gflags.

mosra, 2019-11-26:
mosra/corrade#55 for the file/line info in out is merged now, support is on GCC 8+ and Clang 9+ (and Apple Clang / Xcode 12+, I guess).

mosra, 2020-02-14:
With #496 the file+line info in Debug is available for all GCC versions and Clang 9+.

PR #1411 (2021-08-05): Migrate from glog to ESP's logging
## Motivation and Context

Migration from glog to ESP's logging from #1382.  I also made some thing that previously loud logs on `LOG(INFO)` (now `ESP_DEBUG()`) be verbose logs, so `ESP_VERY_VERBOSE()`.  

The two keys ones are
* URDF logging.  That use to have it's own compile-time flag, but how you can set urdf and physics to VeryVerbose and get those, i.e. use `export HABITAT_SIM_LOG=physics,urdf=VeryVerbose`
* The logging for scene dataset ingestion.  That was just on INFO but is very noisy for large datasets (like hm3d).  That is partially in metadata an [... cut]
closed 2021-08-05 by erikwijmans