facebookresearch/habitat-sim #122: Path issue in setup.py

MarioCavelius, 2019-07-27:
## System

  * Ubuntu 18.04
  * Python 3.6
  * CMake 3.10

## Observed Results

I got the following error when trying to install the simulator. It seems that the main point is that the installer is looking in directory
"/home/mario/Documents/Facebook Habitat/habitat-sim/Habitat/habitat-sim/src" instead of
"/home/mario/Documents/Facebook Habitat/habitat-sim/src".

The installer of release v0.1.1 runs without problems. So it appears that the problem was introduced afterwards. 

```

python setup.py install
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
CMake Error: The source directory "/home/mario/Documents/Facebook Habitat/habitat-sim/Habitat/habitat-sim/src" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Makefile:2040: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Traceback (most recent call last):
  File "setup.py", line 343, in <module>
    zip_safe=False,
  File "/home/mario/Documents/Networks/TensorFlow/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/mario/Documents/Networks/TensorFlow/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/home/mario/Documents/Networks/TensorFlow/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/mario/Documents/Networks/TensorFlow/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/mario/Documents/Networks/TensorFlow/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/mario/Documents/Networks/TensorFlow/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/usr/lib/python3.6/distutils/command/install_lib.py", line 109, in build
    self.run_command('build_ext')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 184, in run
    self.build_extension(ext)
  File "setup.py", line 238, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.
```

erikwijmans, 2019-07-27:
There is a space in the folder name `Facebook Habitat`. Not sure what we changed that would now not be okay with a space but spaces in path names generally cause things to break. Can you try removing it?

MarioCavelius, 2019-07-27:
Thank you! Removing the space solved this issue.
For anyone with the same problem. You need to start with a fresh repo again or fix some stuff happening earlier during the install manually. Just renaming and running it again isn't enough.

closed 2019-07-27 by MarioCavelius