facebookresearch/habitat-sim #127: setup.py install fails on Windows

FishAres, 2019-08-02:
## Steps to reproduce

  1. Clone windows-compatible branch: git checkout -b 2a9deaa
  2. Download requirements: cmake and pip install -r requirements.txt
  3. python setup.py install
  Using python 3.7.3

## Observed Results
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info\PKG-INFO
writing dependency_links to habitat_sim.egg-info\dependency_links.txt
writing requirements to habitat_sim.egg-info\requires.txt
writing top-level names to habitat_sim.egg-info\top_level.txt
reading manifest file 'habitat_sim.egg-info\SOURCES.txt'
writing manifest file 'habitat_sim.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
CMake Error: The source directory "C:UsersaresfCodesim_envshabitat-simsrc" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Traceback (most recent call last):
  File "setup.py", line 343, in <module>
    zip_safe=False,
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\site-packages\setuptools\__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\distutils\core.py", line 148, in setup
    dist.run_commands()
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\distutils\dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\site-packages\setuptools\command\install.py", line 67, in run
    self.do_egg_install()
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\site-packages\setuptools\command\install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\site-packages\setuptools\command\bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\site-packages\setuptools\command\bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\site-packages\setuptools\command\install_lib.py", line 11, in run
    self.build()
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\distutils\command\install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 184, in run
    self.build_extension(ext)
  File "setup.py", line 234, in build_extension
    env=env,
  File "C:\Users\aresf\AppData\Local\Continuum\anaconda3\envs\tf2\lib\subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-HC:UsersaresfCodesim_envshabitat-simsrc', '-BbuildRelease', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=C:\\Users\\aresf\\Code\\sim_envs\\habitat-sim\\build\\lib.win-amd64-3.7\\habitat_sim\\_ext', '-DPYTHON_EXECUTABLE=C:\\Users\\aresf\\AppData\\Local\\Continuum\\anaconda3\\envs\\tf2\\python.exe', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=ON', '-DBUILD_TESTS=OFF']' returned non-zero exit status 1.

erikwijmans, 2019-08-03:
Windows won't be supported until https://github.com/facebookresearch/habitat-sim/pull/84

closed 2020-01-10 by mathfac

Hezhexi2002, 2023-01-15:
> For me unfortunately there still seem to be issues connecting with the GPU from WSL2 when running habitat-sim (headless and not headless). I believe I installed all the right drivers and such. I wonder if someone else succeeded in this!

@Skylion007 Hi,I also run habitat-sim on wsl2 but the simulation is very laggy like the video below:
https://user-images.githubusercontent.com/53631206/212538001-1178de20-c505-4e10-84fe-226bc3e63394.mp4
Actually I was attending the [RoboMaster University Sim2Real Challenge](https://air.tsinghua.edu.cn/robomaster/sim2real_icra23.html) and the organizing committee provide a docker images here for us to test our algorithms,you can see a more detailed introduction [here](https://github.com/AIR-DISCOVER/ICRA-RM-Sim2Real-Baseline).The problem is that although I have setup everything needed to use gpu on wsl2,but the GPU utilization is always almost 0 and the simulation is very laggy too.However,I run the example provided by Microsoft [here](https://learn.microsoft.com/zh-cn/windows/wsl/tutorials/gpu-compute) and the gpu can works well like below：
[image]
[image]
here is the [script ](https://github.com/AIR-DISCOVER/ICRA-RM-Sim2Real-Baseline/blob/master/scripts/launch.sh)provided by the organizing committee to launch the habitat-sim container which is the sim-server shown in the script：
[image]
as you can see,it has already added the `--gpus all` parameter,so I really can't figure out why the gpu is not used during the simulation,I have searched for a long time but diden't find any solution.Hope you can give me some advice:-)