facebookresearch/habitat-sim #129: Build with the provided docker file crash

DianDYu, 2019-08-03:
## Steps to reproduce

  1. Build with the provided docker file
  2. _____
  3. _____

## Observed Results

  * What happened?  This could be a description, log output, etc.
[ 55%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileImporter.cpp.o
[ 55%] Linking CXX static library libMagnumDebugTools.a
[ 55%] Built target MagnumDebugTools
[ 56%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileGen1.cpp.o
[ 56%] Building CXX object esp/geo/CMakeFiles/geo.dir/geo.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [esp/geo/CMakeFiles/geo.dir/CoordinateFrame.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
esp/geo/CMakeFiles/geo.dir/build.make:62: recipe for target 'esp/geo/CMakeFiles/geo.dir/CoordinateFrame.cpp.o' failed
[ 56%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileGen2.cpp.o
[ 56%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileGen3.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
esp/geo/CMakeFiles/geo.dir/build.make:75: recipe for target 'esp/geo/CMakeFiles/geo.dir/OBB.cpp.o' failed
make[2]: *** [esp/geo/CMakeFiles/geo.dir/OBB.cpp.o] Error 4
CMakeFiles/Makefile2:4410: recipe for target 'esp/geo/CMakeFiles/geo.dir/all' failed
make[1]: *** [esp/geo/CMakeFiles/geo.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 56%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/StepExporter.cpp.o
[ 56%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/AMFImporter.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileGen1.cpp.o] Error 4
deps/assimp/code/CMakeFiles/assimp.dir/build.make:816: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileGen1.cpp.o' failed
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1469: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/all' failed
make[1]: *** [deps/assimp/code/CMakeFiles/assimp.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
Traceback (most recent call last):
  File "setup.py", line 140, in <module>
    zip_safe=False,
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 47, in run
    self.build_extension(ext)
  File "setup.py", line 93, in build_extension
    ["cmake", "--build", "."] + build_args, cwd=self.build_temp
  File "/opt/conda/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'RelWithDebInfo', '--', '-j4']' returned non-zero exit status 2.
The command '/bin/sh -c /bin/bash -c ". activate habitat; cd habitat-sim; git checkout 62bc01be8cbc148eb099e96fa0b123e16b21f95f; python setup.py install --headless"' returned a non-zero code: 1
Building docker image failed.

## Expected Results

  * What did you expect to happen?

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

erikwijmans, 2019-08-03:
That happens when you run out of memory when building.  You can reduce the number of workers by changing `-j4` to `-j1` in `setup.py` (more recent versions allow that to be changed with a flag).

DianDYu, 2019-08-05:
Thanks! But I saw the following error when setting number of workers to 1 with commit 954df1fdd661fdda8d6ebc7b91f729e5a09c4b8b:

[ 17%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/DeboneProcess.cpp.o
[ 17%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/ProcessHelper.cpp.o
[ 17%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/MakeVerboseFormat.cpp.o
[ 17%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/ScaleProcess.cpp.o
[ 17%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/MaterialSystem.cpp.o
[ 18%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileImporter.cpp.o
[ 18%] Building CXX object deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileGen1.cpp.o
deps/assimp/code/CMakeFiles/assimp.dir/build.make:816: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileGen1.cpp.o' failed
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [deps/assimp/code/CMakeFiles/assimp.dir/Importer/StepFile/StepFileGen1.cpp.o] Error 4
CMakeFiles/Makefile2:1469: recipe for target 'deps/assimp/code/CMakeFiles/assimp.dir/all' failed
make[1]: *** [deps/assimp/code/CMakeFiles/assimp.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
Traceback (most recent call last):
  File "setup.py", line 343, in <module>
    zip_safe=False,
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/opt/conda/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 184, in run
    self.build_extension(ext)
  File "setup.py", line 238, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/opt/conda/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j1']' returned non-zero exit status 2.
The command '/bin/sh -c /bin/bash -c ". activate habitat; cd habitat-sim; git checkout 954df1fdd661fdda8d6ebc7b91f729e5a09c4b8b; python setup.py build_ext --parallel 1 install --headless"' returned a non-zero code: 1
Building docker image failed.

DianDYu, 2019-08-08:
I ended up using a larger memory to run. Thanks for your help!

closed 2019-08-08 by DianDYu