facebookresearch/habitat-sim #130: Combining 2 meshes into one scene

BurakHocaoglu, 2019-08-05:
Hello there, is there any way to spawn a mesh of an object (e.g. a hammer, a saw) into a scene mesh provided that the appropriate format exists, e.g. via obj2gltf package? If I go with offline spawning, then I think I have to modify scene meshes which is not easy; therefore, if there is an online way (while habitat-sim is running), I'd be glad to have your advice.

erikwijmans, 2019-08-06:
This is something that will be enabled as part of physic integration.  See #91 for the current progress on that front.

BurakHocaoglu, 2019-08-19:
I tested the merged version and I guess that the objects should have a set of physical properties. My aim is to put the objects on random locations; therefore, is it possible that I give restitution, friction, mass etc. small random values just to have a valid format for parsing?

aclegg3, 2019-08-19:
This functionality is now fully supported in Habitat-sim. You can add your own objects in the same way as our example data-set: http://dl.fbaipublicfiles.com/habitat/objects_v0.1.zip. Create a hammer.phys_properties.json file in the objects directory with your mesh file. In that file, create an entry pointing to your mesh file (e.g.): 
{
    "render mesh": "hammer.glb"
}
All other fields can be optionally defined or will be filled with defaults. Center of mass and inertia will be computed automatically if necessary for simulation. At this point, you should create an entry in the global physics config: "hatitat-sim/data/default.phys_scene_config.json". Add your new item to the list of relative paths for rigid objects (e.g.): 
"rigid object paths":[
        "objects/cheezit",
        "objects/chefcan",
        "objects/banana",
        "objects/hammer"
    ] 
At run-time (with --enable_physics) you should see all objects including your being loaded. A call to addObject() with the index of your item in the library should then create one instance of your object in the scene and return it's objectID. Use the Simulator's setTransform(T, objectID) function to modify the object position as desired.

closed 2019-08-23 by aclegg3

BurakHocaoglu, 2019-08-25:
@aclegg3  Thanks for the help. I have more technical question. When I looked at the functions that you mentioned, I saw C++ API relies on transformations of type Magnum::Matrix4. Should I use 2D NumPy arrays of size 4 x 4 in Python API of _Simulator_ object?

aclegg3, 2019-08-25:
@BurakHocaoglu yes, you can update the transforms in python for an object with 2D numpy arrays as in this example:
`T = self._sim.get_transformation(obj_id)
                    print(T)
                    I = np.array([[1.,0,0,0],
                    [0,1.,0,0],
                    [0,0,1.,0],
                    [0,0,0,1.]])
                    self._sim.set_transformation(I, obj_id)
                    T = self._sim.get_transformation(obj_id)
                    print(T)`

You can also using the get/set_translation() and get/set_rotation() methods to directly edit the position (3D vector) or orientation (quaternion) of an object.