facebookresearch/habitat-sim #131: quaternion

jingweiz, 2019-08-06:
Hi, thanks for the repo! 
A quick question, just to be sure, [here](https://aihabitat.org/habitat-api/core/simulator.html#habitat.core.simulator.Simulator.get_observations_at), for ```get_observations_at``` it is stated that ```rotation``` is in the format of ```[x,y,z,w]```; While [here](https://aihabitat.org/habitat-sim/src/agent.html#habitat_sim.SixDOFPose) it is stated that ```rotation``` is in the format of ```np.quaternion```, which should be ```[w,x,y,z]```.
Could you clarify for example for [here](https://github.com/facebookresearch/habitat-sim/blob/0214e2f1bfd43fef2f211c9e7366771d6795075b/examples/demo_runner.py#L122) in the  ```examples/demo_runner.py``` which format is the ```rotation``` in? And which format is used consistently in the rest of the repo? 
Thanks in advance!

erikwijmans, 2019-08-06:
Quaternions represented as lists or numpy arrays are in `[x,y,z,w]` format, quaternions that are `np.quaternion` get printed as `w,x,y,z` format (this is also the format if you use `quaternion.as_float_array` to convert to an array).  Habitat-sim has `habitat_sim.utils.quat_from_coeffs` and `habitat_sim.utils.quat_to_coeffs` to go between `np.quaternion` and a numpy array or list in `[x,y,z,w]` format.  These are used consistently to convert between arrays/lists and `np.quaternion`.

For the example that you pointed to, that is an `np.quaternion`, so that will be printed like `quaternion(w, x, y, z)`.

jingweiz, 2019-08-06:
It is still wierd, I collected a sequence of poses using the poses saved from demo_runner.py, then I try to visualize it using ```quiver```. I converted those saved quaternions using ```habitat_sim.utils.quat_to_angle_axis``` and feed the resulting directional vector to ```quiver``` as ```u,v,w```. However, When the saved quaternions are interpreted as ```w,x,y,z``` the results definately looks wrong, and it looks correct when I transpose the saved quaternions to ```x,y,z,w```. Any help would be appreciated.

erikwijmans, 2019-08-07:
`habitat_sim.utils.quat_to_angle_axis` returns an angle and the axis to rotate about, it does not return a direction vector.  If you want a direction vector, `dir_vec = habitat_sim.utils.quat_rotate_vector(quat, habitat_sim.geo.FRONT)` does that.

jingweiz, 2019-08-07:
Oh that's very helpful! Thanks a lot!

closed 2019-08-07 by jingweiz