facebookresearch/habitat-sim #135: Build from Docker: Segmentation fault (score dumped)

DianDYu, 2019-08-07:
[image]
  1. I tried to build docker with the provided docker file and run "python habitat-sim/examples/example.py" but got "Segmentation fault (core dumped)"
  2. Then I pull and run docker from fairembodied/habitat:latest. I got the following error:

==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0807 06:48:37.864651   392 WindowlessContext.cpp:93] Check failed: numDevices > 0 [EGL] No devices detected
*** Check failure stack trace: ***
Aborted (core dumped)

Any suggestion on how to solve both of this (I would prefer the first one because I can change the docker and build upon it). Thanks!

erikwijmans, 2019-08-07:
I suspect that not being able to pick-up any GPUs and the segfault are related.  Does `nvidia-smi` while inside the docker container produce the expected output (i.e. shows all the systems GPUs)?  EGL also only works with nvidia GPUs on linux AFAIK and I believe this is also true even when inside a docker container (i.e. the host system must be linux).

closed 2019-08-08 by DianDYu

AAdeleye, 2019-12-06:
How did you solve this issue? I have a similar issue:

[image]

The docker file with git checkout version "9129fb2d35569cbdc07ac784fd58d6287240ad8b" for HabitatSim and "55fc9353a81d9a2a899dcd7bc08c8be39d228b8c" for Habitat-api