facebookresearch/habitat-sim #145: compilation error using --headless

UncleMEDM, 2019-08-12:
I am compiling the habitat-sim using command ./build.sh --headless on a remote server. The OS is Ubuntu 16.04.5, gcc version is 5.4.0. During the compilation, I encountered such errors which are related to the EGL library:

[520/544] Building CXX object esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o
FAILED: esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o
/usr/bin/c++  -DGOOGLE_GLOG_DLL_DECL="" -I/data1/haolin/habitat-sim/src/deps -isystem /data1/haolin/habitat-sim/src/cmake/../deps/eigen-git-mirror -isystem /data1/haolin/habitat-sim/src/cmake/../deps/Sophus -isystem /data1/haolin/habitat-sim/src/cmake/../deps/rapidjson/include -isystem /data1/haolin/habitat-sim/src/cmake/../deps/tinyply/source -I/data1/haolin/habitat-sim/src -I/data1/haolin/habitat-sim/src/cmake/../deps/glad/include -I. -I/data1/haolin/habitat-sim/src/deps/corrade/src -Ideps/corrade/src -Ideps/glog -I/data1/haolin/habitat-sim/src/deps/glog/src -I/data1/haolin/habitat-sim/src/deps/magnum/src -Ideps/magnum/src -I/data1/haolin/habitat-sim/src/deps/magnum/src/MagnumExternal/OpenGL -I/data1/haolin/habitat-sim/src/deps/magnum-plugins/src -Ideps/magnum-plugins/src -I/data1/haolin/habitat-sim/src/deps/magnum-integration/src -isystem /usr/local/include -I/data1/haolin/habitat-sim/src/deps/assimp/code/../include -Ideps/assimp/code/../include -DVERSION_INFO=\"0.1.2\" -fvisibility=hidden -O2 -g -DNDEBUG -fPIC   -fopenmp -std=gnu++14 -MD -MT esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o -MF esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o.d -o esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o -c /data1/haolin/habitat-sim/src/esp/gfx/WindowlessContext.cpp
In file included from /data1/haolin/habitat-sim/src/esp/gfx/WindowlessContext.cpp:17:0:
/data1/haolin/habitat-sim/src/esp/gfx/WindowlessContext.cpp: In constructor ‘esp::gfx::{anonymous}::ESPEGLContext::ESPEGLContext(int)’:
/data1/haolin/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:276:43: error: expected primary-expression before ‘,’ token
 #define EGL_NO_CONTEXT EGL_CAST(EGLContext,0)
                                           ^
/data1/haolin/habitat-sim/src/esp/gfx/WindowlessContext.cpp:155:54: note: in expansion of macro ‘EGL_NO_CONTEXT’
     context_ = eglCreateContext(display_, eglConfig, EGL_NO_CONTEXT, NULL);
                                                      ^
/data1/haolin/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:276:45: error: ‘EGL_CAST’ was not declared in this scope
 #define EGL_NO_CONTEXT EGL_CAST(EGLContext,0)
                                             ^
/data1/haolin/habitat-sim/src/esp/gfx/WindowlessContext.cpp:155:54: note: in expansion of macro ‘EGL_NO_CONTEXT’
     context_ = eglCreateContext(display_, eglConfig, EGL_NO_CONTEXT, NULL);
                                                      ^
/data1/haolin/habitat-sim/src/esp/gfx/WindowlessContext.cpp: In member function ‘virtual void esp::gfx::{anonymous}::ESPEGLContext::makeCurrent()’:
/data1/haolin/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:278:43: error: expected primary-expression before ‘,’ token
 #define EGL_NO_SURFACE EGL_CAST(EGLSurface,0)
                                           ^
/data1/haolin/habitat-sim/src/esp/gfx/WindowlessContext.cpp:168:34: note: in expansion of macro ‘EGL_NO_SURFACE’
         eglMakeCurrent(display_, EGL_NO_SURFACE, EGL_NO_SURFACE, context_);
                                  ^
/data1/haolin/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:278:45: error: ‘EGL_CAST’ was not declared in this scope
 #define EGL_NO_SURFACE EGL_CAST(EGLSurface,0)
                                             ^
/data1/haolin/habitat-sim/src/esp/gfx/WindowlessContext.cpp:168:34: note: in expansion of macro ‘EGL_NO_SURFACE’
         eglMakeCurrent(display_, EGL_NO_SURFACE, EGL_NO_SURFACE, context_);
                                  ^
/data1/haolin/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:278:43: error: expected primary-expression before ‘,’ token
 #define EGL_NO_SURFACE EGL_CAST(EGLSurface,0)
                                           ^
/data1/haolin/habitat-sim/src/esp/gfx/WindowlessContext.cpp:168:50: note: in expansion of macro ‘EGL_NO_SURFACE’
         eglMakeCurrent(display_, EGL_NO_SURFACE, EGL_NO_SURFACE, context_);
                                                  ^
[538/544] Linking CXX shared module deps/magnum-bindings/src/python/_magnum.cpython-36m-x86_64-linux-gnu.so
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "setup.py", line 355, in <module>
    zip_safe=False,
  File "/home/haolin/anaconda3/envs/mp3d/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/home/haolin/anaconda3/envs/mp3d/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/haolin/anaconda3/envs/mp3d/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/haolin/anaconda3/envs/mp3d/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 190, in run
    self.build_extension(ext)
  File "setup.py", line 250, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/home/haolin/anaconda3/envs/mp3d/lib/python3.6/subprocess.py", line 291, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--']' returned non-zero exit status 1.

Is there any solutions for this? Thank you!

erikwijmans, 2019-08-14:
IIRC, this happens when you have an old `libegl1-mesa-dev`.  On ubuntu/debian, try `sudo apt install libegl1-mesa-dev`

devendrachaplot, 2019-08-14:
I already tried that:
> libegl1-mesa-dev is already the newest version

erikwijmans, 2019-08-14:
Perhaps the OS version doesn't have the most recent version available.  The file that defines EGL_CAST is `EGL/eglplatform.h`, here is the most recent version of it: [khronos.org/registry/EGL/api/EGL/eglplatform.h](https://www.khronos.org/registry/EGL/api/EGL/eglplatform.hl). It should be fine to just put that somewhere in the include path (try `src/EGL`).

erikwijmans, 2019-08-14:
You can also try adding something like
```c++
undef EGL_CAST
#define EGL_CAST(type, value) (static_cast<type>(value))
```
here: https://github.com/facebookresearch/habitat-sim/blob/master/src/esp/core/esp.h#L12 if all else fails.

UncleMEDM, 2019-08-19:
I found that the compilation still fails with the same issue for the previous two methods. For the last method, the compilation was successful, but the program cannot be ran with segmentation fault. Therefore, I decided to install the Habitat using docker.

mathfac, 2019-09-23:
@UncleMEDM, did you succeed to install Habitat Sim? 
We have [an example how we install Habitat-sim in clean Ubuntu in our CI setup](https://github.com/facebookresearch/habitat-sim/blob/master/.circleci/config.yml#L64).
Make sure next dependencies are installed:
```
sudo apt-get update || true
sudo apt-get install -y --no-install-recommends \
                  build-essential \
                  git \
                  curl \
                  vim \
                  ca-certificates \
                  libjpeg-dev \
                  libglm-dev \
                  libegl1-mesa-dev \
                  xorg-dev \
                  freeglut3-dev \
                  pkg-config \
                  wget \
                  zip \
                  unzip
```

UncleMEDM, 2019-09-24:
@mathfac I have tried that before, but it fails in my original Ubuntu 16.04. When I reinstalled my computer with Ubuntu 18.04, it works for the compilation.

closed 2019-11-22 by mathfac

BryanBo-Cao, 2020-05-29:
Curious if default installation option install cuda features automatically when GPU is available.

I have installed CUDA 10.2 and nvidia drivers manually for my GPU in my Alienware laptop 17 R5(newly used):
```
Fri May 29 18:49:34 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.59       Driver Version: 440.59       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1070    Off  | 00000000:01:00.0  On |                  N/A |
| N/A   48C    P0    35W /  N/A |    543MiB /  8117MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1289      G   /usr/lib/xorg/Xorg                            18MiB |
|    0      1423      G   /usr/bin/gnome-shell                          48MiB |
|    0      1740      G   /usr/lib/xorg/Xorg                           190MiB |
|    0      1856      G   /usr/bin/gnome-shell                         199MiB |

```
and try to build everything of habitat.

```
python setup.py install --with-cuda
```
gave me an error but I built successfully by
```
python setup.py install
```