facebookresearch/habitat-sim #146: Build failed inside a singularity container

devendrachaplot, 2019-08-13:
I am trying to build habitat inside a singularity container. I am getting the following error:

## Steps to reproduce
  > singularity shell -w image_name.img
  git clone https://github.com/facebookresearch/habitat-sim; cd habitat-sim
  pip install -r requirements.txt
  ./build.sh --headless

## Observed Results
>[435/585] Building CXX object esp/core/CMakeFiles/core.dir/esp.cpp.o
FAILED: esp/core/CMakeFiles/core.dir/esp.cpp.o
/usr/bin/c++  -DGOOGLE_GLOG_DLL_DECL="" -I/root/habitat-sim/src/deps -I/root/habitat-sim/src -I/root/habitat-sim/src/deps/corrade/src -Ideps/corrade/src -Ideps/glog -I/root/habitat-sim/src/deps/glog/src -isystem /root/habitat-sim/src/cmake/../deps/eigen-git-mirror -isystem /root/habitat-sim/src/cmake/../deps/Sophus -
isystem /root/habitat-sim/src/cmake/../deps/tinyobjloader -isystem /root/habitat-sim/src/cmake/../deps/rapidjson/include -isystem /root/habitat-sim/src/cmake/../deps/assimp/include -isystem /root/habitat-sim/src/cmake/../deps/googletest/googletest/include -fvisibility=hidden -O2 -g -DNDEBUG -fPIC   -std=gnu++14 -MD -
MT esp/core/CMakeFiles/core.dir/esp.cpp.o -MF esp/core/CMakeFiles/core.dir/esp.cpp.o.d -o esp/core/CMakeFiles/core.dir/esp.cpp.o -c /root/habitat-sim/src/esp/core/esp.cpp
In file included from /root/habitat-sim/src/deps/eigen-git-mirror/Eigen/Core:143:0,
                 from /root/habitat-sim/src/esp/core/esp.h:17,
                 from /root/habitat-sim/src/esp/core/esp.cpp:5:
/root/habitat-sim/src/deps/eigen-git-mirror/Eigen/src/Core/util/IntegralConstant.h:189:36: error: template declaration of ‘const Eigen::internal::FixedInt<N> Eigen::fix’
 static const internal::FixedInt<N> fix{};
                                    ^
In file included from /root/habitat-sim/src/deps/eigen-git-mirror/Eigen/Core:224:0,
                 from /root/habitat-sim/src/esp/core/esp.h:17,
                 from /root/habitat-sim/src/esp/core/esp.cpp:5:
/root/habitat-sim/src/deps/eigen-git-mirror/Eigen/src/Core/util/IndexedViewHelper.h:57:147: error: ‘fix’ was not declared in this scope
 static const symbolic::AddExpr<symbolic::SymbolExpr<internal::symbolic_last_tag>,symbolic::ValueExpr<Eigen::internal::FixedInt<1> > > lastp1(last+fix<1>());
                                                                                                                                                   ^
/root/habitat-sim/src/deps/eigen-git-mirror/Eigen/src/Core/util/IndexedViewHelper.h:57:154: error: expected primary-expression before ‘)’ token
 static const symbolic::AddExpr<symbolic::SymbolExpr<internal::symbolic_last_tag>,symbolic::ValueExpr<Eigen::internal::FixedInt<1> > > lastp1(last+fix<1>());
                                                                                                                                                          ^
In file included from /root/habitat-sim/src/deps/eigen-git-mirror/Eigen/Core:226:0,
                 from /root/habitat-sim/src/esp/core/esp.h:17,
                 from /root/habitat-sim/src/esp/core/esp.cpp:5:
/root/habitat-sim/src/deps/eigen-git-mirror/Eigen/src/Core/ArithmeticSequence.h:142:64: error: ‘fix’ was not declared in this scope
   auto reverse() const -> decltype(Eigen::seqN(m_first+(m_size+fix<-1>())*m_incr,m_size,-m_incr)) {
                                                                ^
/root/habitat-sim/src/deps/eigen-git-mirror/Eigen/src/Core/ArithmeticSequence.h:142:64: error: ‘fix’ was not declared in this scope
/root/habitat-sim/src/deps/eigen-git-mirror/Eigen/src/Core/ArithmeticSequence.h:142:72: error: expected primary-expression before ‘)’ token
   auto reverse() const -> decltype(Eigen::seqN(m_first+(m_size+fix<-1>())*m_incr,m_size,-m_incr)) {

Any pointers to what might be causing this issue?

erikwijmans, 2019-08-13:
What compiler/version are you using? That seems like a compiler error.

devendrachaplot, 2019-08-14:
Do you mean g++ version?

> g++ --version
g++ (Ubuntu 4.9.4-2ubuntu1~14.04.1) 4.9.4

devendrachaplot, 2019-08-14:
I was able to get past the previous error by updating g++ version.

Now I am getting some EGL error:
```
[496/544] Building CXX object esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o
FAILED: esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o
/usr/bin/c++  -DGOOGLE_GLOG_DLL_DECL="" -I/root/habitat-sim/src/deps -I/root/habitat-sim/src -I/root/habitat-sim/src/cmake/../deps/glad/include -I. -I/root/habitat-sim/src/deps/corrade/src -Ideps/corrade/src -Ideps/glog -I/root/habitat-sim/src/deps/glog/src -I/root/habitat-sim/
src/deps/magnum/src -Ideps/magnum/src -I/root/habitat-sim/src/deps/magnum/src/MagnumExternal/OpenGL -I/root/habitat-sim/src/deps/magnum-plugins/src -Ideps/magnum-plugins/src -I/root/habitat-sim/src/deps/magnum-integration/src -I/root/habitat-sim/src/deps/assimp/code/../include
-Ideps/assimp/code/../include -isystem /root/habitat-sim/src/cmake/../deps/eigen-git-mirror -isystem /root/habitat-sim/src/cmake/../deps/Sophus -isystem /root/habitat-sim/src/cmake/../deps/rapidjson/include -isystem /root/habitat-sim/src/cmake/../deps/tinyply/source -DVERSION_I
NFO=\"0.1.2\" -fvisibility=hidden -O2 -g -DNDEBUG -fPIC   -fopenmp -std=gnu++14 -MD -MT esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o -MF esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o.d -o esp/gfx/CMakeFiles/gfx.dir/WindowlessContext.cpp.o -c /root/habitat-sim/src/esp
/gfx/WindowlessContext.cpp
In file included from /root/habitat-sim/src/esp/gfx/WindowlessContext.cpp:17:0:
/root/habitat-sim/src/esp/gfx/WindowlessContext.cpp: In constructor ‘esp::gfx::{anonymous}::ESPEGLContext::ESPEGLContext(int)’:
/root/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:276:43: error: expected primary-expression before ‘,’ token
 #define EGL_NO_CONTEXT EGL_CAST(EGLContext,0)
                                           ^
/root/habitat-sim/src/esp/gfx/WindowlessContext.cpp:155:54: note: in expansion of macro ‘EGL_NO_CONTEXT’
     context_ = eglCreateContext(display_, eglConfig, EGL_NO_CONTEXT, NULL);
                                                      ^
/root/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:276:45: error: ‘EGL_CAST’ was not declared in this scope
 #define EGL_NO_CONTEXT EGL_CAST(EGLContext,0)
                                             ^
/root/habitat-sim/src/esp/gfx/WindowlessContext.cpp:155:54: note: in expansion of macro ‘EGL_NO_CONTEXT’
     context_ = eglCreateContext(display_, eglConfig, EGL_NO_CONTEXT, NULL);
                                                      ^
/root/habitat-sim/src/esp/gfx/WindowlessContext.cpp: In member function ‘virtual void esp::gfx::{anonymous}::ESPEGLContext::makeCurrent()’:
/root/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:278:43: error: expected primary-expression before ‘,’ token
 #define EGL_NO_SURFACE EGL_CAST(EGLSurface,0)
                                           ^
/root/habitat-sim/src/esp/gfx/WindowlessContext.cpp:168:34: note: in expansion of macro ‘EGL_NO_SURFACE’
         eglMakeCurrent(display_, EGL_NO_SURFACE, EGL_NO_SURFACE, context_);
                                  ^
/root/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:278:45: error: ‘EGL_CAST’ was not declared in this scope
 #define EGL_NO_SURFACE EGL_CAST(EGLSurface,0)
                                             ^
/root/habitat-sim/src/esp/gfx/WindowlessContext.cpp:168:34: note: in expansion of macro ‘EGL_NO_SURFACE’
         eglMakeCurrent(display_, EGL_NO_SURFACE, EGL_NO_SURFACE, context_);
                                  ^
/root/habitat-sim/src/cmake/../deps/glad/include/glad/glad_egl.h:278:43: error: expected primary-expression before ‘,’ token
 #define EGL_NO_SURFACE EGL_CAST(EGLSurface,0)
                                           ^
/root/habitat-sim/src/esp/gfx/WindowlessContext.cpp:168:50: note: in expansion of macro ‘EGL_NO_SURFACE’
         eglMakeCurrent(display_, EGL_NO_SURFACE, EGL_NO_SURFACE, context_);
                                                  ^
[509/544] Linking CXX shared module deps/magnum-bindings/src/python/_magnum.cpython-36m-x86_64-linux-gnu.so
ninja: build stopped: subcommand failed.
```

erikwijmans, 2019-08-14:
IIRC, this happens when you have an old `libegl1-mesa-dev`.  Try running `sudo apt install libegl1-mesa-dev`

devendrachaplot, 2019-08-14:
I already tried that:
> libegl1-mesa-dev is already the newest version.

msbaines, 2019-08-19:
I've created a patch to fix this. Please try it and let me know if it resolves this issue.

To try:

```
git fetch origin
git cherry-pick origin/ubuntu
```

BurakHocaoglu, 2019-09-26:
> I already tried that:
> 
> > libegl1-mesa-dev is already the newest version.

When I checked the available packages for EGL, I have found several that may be related:
* libegl1-mesa-dev (already mentioned)
* libegl1-mesa-drivers
* libegl1-mesa-lts-xenial (for Ubuntu 16.04, I believe)

Installing all 3 of them solved my EGL build issue.

devendrachaplot, 2019-09-28:
I created a singularity container by converting the docker image and that worked.

closed 2019-09-28 by devendrachaplot