facebookresearch/habitat-sim #167: Seg fault on OSX when initializing with a glb file

dhuber23b, 2019-08-20:
## Steps to reproduce

  1. Run the attached script, which reproduces the problem.

## Observed Results

Program segfaults with no error message.

## Expected Results

No seg fault.  If you run with a sample scene from the sample data, it will work fine.

## Relevant Code

```
import habitat_sim

def test(scene_path):
    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.gpu_device_id = 0
    sim_cfg.scene.id = scene_path
        
    sensor_spec = habitat_sim.SensorSpec()
    sensor_spec.uuid = "color_sensor"
    sensor_spec.sensor_type = habitat_sim.SensorType.COLOR
    sensor_spec.resolution = [256, 256]
    sensor_spec.position = [0, 0.5, 0]
    sensor_specs = [sensor_spec]    
    
    agent_cfg = habitat_sim.agent.AgentConfiguration()
    agent_cfg.sensor_specifications = sensor_specs
    agent_cfg.action_space = {
        "move_forward": habitat_sim.agent.ActionSpec(
            "move_forward", habitat_sim.agent.ActuationSpec(amount=0.25)
        ),
    }
    cfg = habitat_sim.Configuration(sim_cfg, [agent_cfg])        
    sim = habitat_sim.Simulator(cfg)

    # set up agent 0
    sim.initialize_agent(0)
    
scene_name = "coffee_mug_scaled.glb"
sim = test(scene_name)    
[crash_habitat.zip](https://github.com/facebookresearch/habitat-sim/files/3522738/crash_habitat.zip)

```

erikwijmans, 2019-08-21:
`sim.initialize_agent(0)` relies on the navigation mesh to find a position for the agent, but, I would assume, the coffee cup doesn't have a navmesh, so that's why its segfaulting.  It shouldn't segfault, but I am not quite sure what it should do in this situation.  There isn't a logical position to set the agent to -- should it set the agent to n arbitrary position and heading, pos = [0, 0, 0]  facing along the -z axis,  throw an error, or do nothing?

dhuber23b, 2019-08-21:
Erik, is there some documentation about the navmesh format or how it works?  Maybe I can create one.  Interestingly, the viewer app in c++ works ok without this extra mesh.  

I like your idea of initializing the agent to 0,0,0 facing -z.

aclegg3, 2019-08-21:
Hey Daniel, functionality is live on master now to dynamically add objects to scenes with kinematic manipulation in both C++ and python. In this case you could load any existing scene, load the "coffee_mug_scaled.glb" file into the object library, and then initialize it within the scene and move it wherever you want it. Would this work better for you, or is there a specific reason you would like to use the mug as a scene mesh? Happy to help in either case.

erikwijmans, 2019-08-21:
Most of habitat is fine without the navmesh, but it seems like we forgot to the make the initialize method do something sensible when there isn't a navmesh.

PR #169 (2019-08-22): Fix not having a navmesh
## Motivation and Context

Fixes #167 

## How Has This Been Tested

With the test

## Types of changes

- Bug fix (non-breaking change which fixes an issue)
closed 2019-08-22 by msavva