facebookresearch/habitat-sim #179: Error Undefined Symbol for habitat_sim_bindings

SenZHANG-GitHub, 2019-08-30:
## Steps to reproduce

  1. CentOS release 6.9 (Linux version 2.6.32-696.16.1.el6.x86_64 (gcc version 4.4.7 20120313 (Red Hat 4.4.7-18) (GCC) )
  2. Complier: gcc 7.2.0
  3. Compilation status: success (I suppose)
  4. python version: 3.6.5 (GCC 4.9.3 on linux) (I add export /usr/local/gcc/7.2.0/lib64/libstdc++.so.6 to fix a runtime issue w.r.t incompatible old versions of libstdc++.so.6)

## Observed Results

Runtime error when I tried to impose habitat_sim. Could you give me some suggestions to fix this?

(habitat) szha2609@login4:~/src$ py
Python 3.6.5 (default, May  3 2018, 10:16:57)
[GCC 4.9.3] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import habitat_sim
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/szha2609/venv/habitat/lib/python3.6/site-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/__init__.py", line 12, i
n <module>
    from .nav import *
  File "/home/szha2609/venv/habitat/lib/python3.6/site-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/nav/__init__.py", line 1
, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/home/szha2609/venv/habitat/lib/python3.6/site-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/nav/greedy_geodesic_foll
ower.py", line 6, in <module>
    import habitat_sim.agent
  File "/home/szha2609/venv/habitat/lib/python3.6/site-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/agent/__init__.py", line
 7, in <module>
    from .agent import *
  File "/home/szha2609/venv/habitat/lib/python3.6/site-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/agent/agent.py", line 14
, in <module>
    import habitat_sim.bindings as hsim
  File "/home/szha2609/venv/habitat/lib/python3.6/site-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/bindings/__init__.py", l
ine 24, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ImportError: /home/szha2609/venv/habitat/lib/python3.6/site-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/_ext/habitat_sim_bi
ndings.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6google14FlagRegistererC1INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEPKcS9_S9_PT_SB_

## Relevant Code

Compilation Information is attached in compilation_info.txt.

[compilation_info.txt](https://github.com/facebookresearch/habitat-sim/files/3558681/compilation_info.txt)

---

SenZHANG-GitHub, 2019-08-30:
"ldd /home/szha2609/venv/habitat/lib/python3.6/site-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/_ext/habitat_sim_bi
ndings.cpython-36m-x86_64-linux-gnu.so" gives the following results. Anything missing or incorrect?

        linux-vdso.so.1 =>  (0x00007ffcf38d6000)
	/usr/local/gcc/7.2.0/lib64/libstdc++.so.6 (0x00002b8f61567000)
	libz.so.1 => /usr/local/zlib/1.2.8/lib/libz.so.1 (0x00002b8f618fe000)
	librt.so.1 => /lib64/librt.so.1 (0x00002b8f61b2e000)
	libgomp.so.1 => /usr/local/gcc/7.2.0/lib64/libgomp.so.1 (0x00002b8f61d36000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00002b8f61f64000)
	libGL.so.1 => /usr/lib64/libGL.so.1 (0x00002b8f62181000)
	libEGL.so.1 => /home/szha2609/usr/lib64/libEGL.so.1 (0x00002b8f6240a000)
	libCorradePluginManager.so.2 => /home/szha2609/src/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext/libCorradePluginManager.so.2 (0x00002b8f6262d000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00002b8f6283e000)
	libCorradeUtility.so.2 => /home/szha2609/src/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext/libCorradeUtility.so.2 (0x00002b8f62a42000)
	libgflags.so.2.2 => /usr/local/gflags/2.2.1/lib/libgflags.so.2.2 (0x00002b8f62c8f000)
	libm.so.6 => /lib64/libm.so.6 (0x00002b8f62ec8000)
	libgcc_s.so.1 => /usr/local/gcc/7.2.0/lib64/libgcc_s.so.1 (0x00002b8f6314c000)
	libc.so.6 => /lib64/libc.so.6 (0x00002b8f63363000)
	/lib64/ld-linux-x86-64.so.2 (0x0000003dc4600000)
	libglapi.so.0 => /usr/lib64/libglapi.so.0 (0x00002b8f636f7000)
	libselinux.so.1 => /lib64/libselinux.so.1 (0x00002b8f63925000)
	libXext.so.6 => /usr/lib64/libXext.so.6 (0x00002b8f63b45000)
	libXdamage.so.1 => /usr/lib64/libXdamage.so.1 (0x00002b8f63d57000)
	libXfixes.so.3 => /usr/lib64/libXfixes.so.3 (0x00002b8f63f59000)
	libX11-xcb.so.1 => /usr/lib64/libX11-xcb.so.1 (0x00002b8f6415f000)
	libX11.so.6 => /usr/lib64/libX11.so.6 (0x00002b8f64360000)
	libxcb-glx.so.0 => /usr/lib64/libxcb-glx.so.0 (0x00002b8f6469d000)
	libxcb-dri2.so.0 => /usr/lib64/libxcb-dri2.so.0 (0x00002b8f648b6000)
	libxcb.so.1 => /usr/lib64/libxcb.so.1 (0x00002b8f64aba000)
	libXxf86vm.so.1 => /usr/lib64/libXxf86vm.so.1 (0x00002b8f64cdf000)
	libdrm.so.2 => /usr/lib64/libdrm.so.2 (0x00002b8f64ee5000)
	libexpat.so.1 => /lib64/libexpat.so.1 (0x00002b8f650f2000)
	libxcb-xfixes.so.0 => /usr/lib64/libxcb-xfixes.so.0 (0x00002b8f6531a000)
	libxcb-render.so.0 => /usr/lib64/libxcb-render.so.0 (0x00002b8f65522000)
	libxcb-shape.so.0 => /usr/lib64/libxcb-shape.so.0 (0x00002b8f6572e000)
	libgbm.so.1 => /usr/lib64/libgbm.so.1 (0x00002b8f65931000)
	libXau.so.6 => /usr/lib64/libXau.so.6 (0x00002b8f65b3d000)

mosra, 2019-08-30:
The project uses, IIRC, bundled gflags (used by glog I think), and in your case it seems like it compiled with the bundled version but then is trying to use the system-wide library from `/usr/local/gflags/2.2.1/lib/libgflags.so.2.2` and those two are different, causing the missing symbol failure. Or maybe, due to how you preload the other `libstdc++.so`, it also accidentally picked up a different gflags version?

Is there any `libgflags.so` anywhere else on your system? Can you try running with the `LD_LIBRARY_PATH` environment variable set to that location? Or `LD_PRELOAD` that library?

SenZHANG-GitHub, 2019-09-02:
> The project uses, IIRC, bundled gflags (used by glog I think), and in your case it seems like it compiled with the bundled version but then is trying to use the system-wide library from [...]

Should be exactly the problem. I previously add -Dgflags_DIR=/usr/local/gflags/2.2.1/lib/cmake/gflags to setup.py to avoid the following warning:
---
CMake Warning at deps/glog/CMakeLists.txt:49 (find_package):
  By not providing "Findgflags.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gflags", but
  CMake did not find one.

  Could not find a package configuration file provided by "gflags" (requested
  version 2.2.0) with any of the following names:

    gflagsConfig.cmake
    gflags-config.cmake

  Add the installation prefix of "gflags" to CMAKE_PREFIX_PATH or set
  "gflags_DIR" to a directory containing one of the above files.  If "gflags"
  provides a separate development package or SDK, be sure it has been
  installed.
---
Now I just remove -Dgflags_DIR=/usr/local/gflags/2.2.1/lib/cmake/gflags  and recompile regardless of this warning. The compilation still succeed and get no problem to import habitat_sim now. Though another issue occurs when I run example.py (Segmentation fault (core dumped)). Trying to identify the problem and reading other issues now. 

The recompilation log is attached for anyone interested in it. 

[wo_gflags_compilation.log](https://github.com/facebookresearch/habitat-sim/files/3564220/wo_gflags_compilation.log)

closed 2019-09-02 by SenZHANG-GitHub

mosra, 2019-09-25:
Just FYI, this is being fixed on our side in #247.

SenZHANG-GitHub, 2019-09-26:
> Just FYI, this is being fixed on our side in #247.

Thanks!