facebookresearch/habitat-sim #180: Installation  Error

meikai1023, 2019-08-31:
I'm very interested in habitat，I've been trying for maney days,first on wendows10 and it faild,
now I am trying  on linux but it also faild ，I've done  what i can ,please help me!

Environment:
  1. ubuntu16.04 
  2. gcc 7.1.0
  3. cmake 3.14.4
  4. ptyhon3.6

input " python setup.py install  " or "python setup.py install --headless"

output: 

running install
Checking .pth file support in /usr/local/lib/python3.6/dist-packages/
/usr/bin/python3 -E -c pass
TEST PASSED: /usr/local/lib/python3.6/dist-packages/ appears to support .pth files
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at cmake/dependencies.cmake:13 (add_subdirectory):
  The source directory

    /home/lk/opt/habitat-sim-master/src/deps/corrade

  does not contain a CMakeLists.txt file.
Call Stack (most recent call first):
  CMakeLists.txt:45 (include)

CMake Error at /usr/local/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Corrade (missing: CORRADE_INCLUDE_DIR
  _CORRADE_CONFIGURE_FILE)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindCorrade.cmake:286 (find_package_handle_standard_args)
  cmake/dependencies.cmake:15 (find_package)
  CMakeLists.txt:45 (include)

-- Configuring incomplete, errors occurred!
See also "/home/lk/opt/habitat-sim-master/build/CMakeFiles/CMakeOutput.log".
Traceback (most recent call last):
  File "setup.py", line 368, in <module>
    zip_safe=False,
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 161, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 147, in call_command
    self.run_command(cmdname)
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/install_lib.py", line 23, in run
    self.build()
  File "/usr/lib/python3.6/distutils/command/install_lib.py", line 109, in build
    self.run_command('build_ext')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 195, in run
    self.build_extension(ext)
  File "setup.py", line 259, in build_extension
    env=env,
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-H/home/lk/opt/habitat-sim-master/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/lk/opt/habitat-sim-master/build/lib.linux-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/usr/bin/python3', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=OFF', '-DTARGET_HEADLESS=ON', '-DBUILD_TEST=OFF', '-DWITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF']' returned non-zero exit status 1.

erikwijmans, 2019-09-02:
We rely on a number of git-submodules.  You need to clone the repository, not download a zip/tarball for things to work.

closed 2019-09-02 by erikwijmans

meikai1023, 2019-09-03:
> We rely on a number of git-submodules. You need to clone the repository, not download a zip/tarball, for things to work.

thanks。